micropython/micropython #6576: Uasyncio Uart Stream Close Bug

gtomassi, 2020-10-28:
When calling aclose (maps to Stream.wait_closed) on the new implementation of Stream, something goes awry in the stream read process and begins raising exceptions.

**After calling aclose the following stack trace is seen:**
```python
  File "uasyncio/stream.py", line 30, in read
  File "uasyncio/core.py", line 94, in queue_read
  File "uasyncio/core.py", line 82, in _enqueue
```

**Steps to reproduce:**

1. Open a UART connection using the stream reader 'streamReader = asyncio.StreamReader(uart)'
2. Close the stream reader
3. Not sure if this is required, but open a new stream reader on the same UART again

gtomassi, 2020-10-29:
After further investigation, the issue appears to be because the the uart object being passed into the stream reader always has the same unique object ID (id(uart)). So when the old read never releases (and hangs forever), and a new read is requested, the _enqueue operation already has a task waiting on that stream. Thus it throws an assertion error since there shouldn't be a task that already exists for that stream. 

This may actually be a bug in the uart code rather than the new uasyncio, if uart.deinit() does not properly release a read operation.

dpgeorge, 2021-04-14:
I could not reproduce this problem.  I used the following code on a pyboard:
```python
import machine, uasyncio as asyncio

async def main():
    uart = machine.UART('XA', 9600)
    sr = asyncio.StreamReader(uart)
    sr.write('test')
    await sr.drain()
    print('read:', await sr.read(4))
    sr.close()
    await sr.wait_closed()
    
asyncio.run(main())
```
This raises an exception, but not as described above.  Instead it gives:
```
read: b'test'
Traceback (most recent call last):
  File "<stdin>", line 12, in <module>
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 10, in main
  File "uasyncio/stream.py", line 1, in wait_closed
AttributeError: 'UART' object has no attribute 'close'
```

gtomassi, 2021-05-12:
@dpgeorge 
Doesn't seem to happen for us anymore with the latest MP (1.15). So something must've changed, not sure what. Will continue to monitor and reopen if needed.

closed 2021-05-12 by gtomassi

dpgeorge, 2021-05-13:
Great, thanks for letting us know.