micropython/micropython #6582: __class__ bug in ESP32/ESP8266

shaoziyang, 2020-10-29:
When import a module (such as import os), then input "os.__class__.", if press tab key, it will cause ESP32/ESP8266 reset. STM32 has no such problem.

`import os

os.__class__.`

robert-hh, 2020-10-29:
Which firmware version do you use? I do not see such a behavior on the actual build.
Tested with MicroPython v1.13-144-gc7d3861cb on 2020-10-29

jimmo, 2020-10-29:
This is probably #6493 which was fixed recently. Please try using the latest firmware build from micropython.org/download and see if that solves the problem.

shaoziyang, 2020-10-29:
> Which firmware version do you use? I do not see such a behavior on the actual build.
> Tested with MicroPython v1.13-144-gc7d3861cb on 2020-10-29

I use ESP8266/ESP32 firmware stable v1.13.

dpgeorge, 2021-04-14:
This was fixed in v1.14.

closed 2021-04-14 by dpgeorge