micropython/micropython #6583: esp32: uctypes array with int64 not supported?

pumelo, 2020-10-29:
this is experienced on the esp32 port

```
=== import uctypes
===
=== b = bytearray(8+8*50)
=== OUT_STRUCT = {'s': (0 | uctypes.INT64),
===               't': (8 | uctypes.ARRAY, 50 | uctypes.INT64)}
=== d_out = uctypes.struct(uctypes.addressof(b),
===                             OUT_STRUCT)
===
=== d_out.s = 111
=== print(d_out.s)
=== d_out.t[0] = 222
=== print(d_out.t[0])
=== d_out.t[1] = 333
=== print(d_out.t[1])
=== d_out.t[49] = 444
=== print(d_out.t[49])
111
222
0
0
```

dpgeorge, 2020-10-29:
Thanks for the report.  I didn't test this but the following might fix it:
```diff
diff --git a/extmod/moductypes.c b/extmod/moductypes.c
index c5fbf12e4..a8dbfdc4f 100644
--- a/extmod/moductypes.c
+++ b/extmod/moductypes.c
@@ -399,7 +399,7 @@ STATIC void set_aligned(uint val_type, void *p, mp_int_t index, mp_obj_t val) {
                 ((uint64_t *)p)[index] = (uint64_t)v;
             } else {
                 // TODO: Doesn't offer atomic store semantics, but should at least try
-                set_unaligned(val_type, p, MP_ENDIANNESS_BIG, val);
+                set_unaligned(val_type, &((uint64_t *)p)[index], MP_ENDIANNESS_BIG, val);
             }
             return;
         default:
```

pumelo, 2020-11-02:
@dpgeorge Thank you very much for the fast fix.

Just tested this with the above simple example on a esp32: Yes, it works! 💐

Is this a port specific solution or will you merge it?

dpgeorge, 2020-11-05:
Thanks for testing!  This is a general issue (not esp32 specific) that is fixed by #6602.

PR #6602 (2020-11-11): Fix some sign extension bugs, and uctypes array with (U)INT64 elements on 32-bit archs
Trying to fix #6583 and write tests for it I found a few sign extension bugs, which are fixed here along with the uctypes array index bug.
closed 2020-11-11 by dpgeorge