micropython/micropython #6585: native modules not properly rebuilt on ARCH change

agners, 2020-10-30:
When rebuilding a different arch, a unusable module gets built.

```
cd examples/natmod/features1
<edit Makefile to ARCH = xtensawin>
make
<edit Makefile to ARCH = x64>
make
```
Loading this module crashes MicroPython (Unix port)

```
$ ./micropython                                                                          
MicroPython v1.13-148-ged7ddd4dd on 2020-10-30; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import features1
initialising module self=7f709249c620
[1]    2093092 segmentation fault (core dumped)  ./micropython
```

Using a `make clean` before building the x64 module builds a proper module. Maybe a won't fix? Just noticed it...

stinos, 2020-10-30:
There are quite a lot of other cases where rebuilding is necessary after changing build flags, it's probably my number 1 gripe with building MicroPython with Makefiles.

I've been thinking about a fix though; the Makefile already does dependency tracking of include files, which is a start, but not sufficient. I think the solution is to have a build write it's CFLAGS and LDFLAGS to a file (eventually per-file because CFLAGS can be different per file), and when that changes do a rebuild (or build the files affected by these new flags only). That's basically how e.g. msbuild does it as well. Once I figure out how to do that in a Makefile I'll make a PR.

dlech, 2020-10-30:
A workaround that should be possible now is to also provide a different `BUILD` directory as well for different architectures.

dpgeorge, 2020-11-05:
> A workaround that should be possible now is to also provide a different `BUILD` directory as well for different architectures.

Yes, that would be a good solution.