micropython/micropython #6589: uos.uname ().nodename should report dchp_hostname if set, need way to specify node.

tombelpasso, 2020-10-31:
The nodename returned by uname seems to be hard coded.
To be compatible with RPi and PC it should specify the network hostname.

There should be a way to change the nodename and use that as default for network hostname.

Test code. 
>>> from uos import uname
>>> import network
>>> uname()
(sysname='esp32', nodename='esp32', release='1.13.0', version='v1.13 on 2020-09-02', machine='ESP32 module with ESP32')

>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> wlan.connect(ssid, password)

>>> wlan.config(dhcp_hostname="lora32-01")
>>> wlan.config('dhcp_hostname')
'lora32-01'

# NOTE: nodename should be same as dhcp_hostname: 'lora32-01'
>>> uname()
(sysname='esp32', nodename='esp32', release='1.13.0', version='v1.13 on 2020-09-02', machine='ESP32 module with ESP32')

Thanks

dpgeorge, 2020-11-05:
> The nodename returned by uname seems to be hard coded.

Yes, that's because all of the `uname()` result is hard-coded, for efficiency.

> There should be a way to change the nodename and use that as default for network hostname

CPython does not let the `nodename` entry be changed, so that's probably not the right way to go here.

In the end I don't see this as a big issue (but maybe I'm mistaken?) and one can just use `wlan.config('dhcp_hostname')` to get (and set) the network name.

jonnor, 2025-10-03:
Seems there is a solution for this.

closed 2025-10-03 by jonnor