micropython/micropython #6590: Nucleo F429ZI UART

micropyt, 2020-11-01:
I need to change uart pins in firmware. After rewrite in the mpconfigboard.h, rebuild the firmware and call the uart the response is uart 2 doesn't exist. What I have to change to it works?

Thanks for your responses.

dpgeorge, 2020-11-05:
To get help on this, please post your changes to the `NUCLEO_F429ZI` board definition, and the REPL session that shows the error.

micropyt, 2020-11-05:
Here are the posts. The first is the original settings of mpconfigboard.h.

[image]

changed
[image]

the errors
[image]

micropyt, 2020-11-05:
I need to run Modbus with this shield:

https://www.waveshare.com/wiki/RS485_CAN_Shield

micropyt, 2020-11-12:
Is possible to use the Pin alternate function on stm32?

davehylands, 2020-11-13:
Yes.

These types of questions are best asked on the [forum](https://forum.micropython.org/). github issues are for tracking bugs.

micropyt, 2020-11-13:
Thanks, but I have to say, that it doesn't work probably.
UART does not work on the new pins.

dpgeorge, 2021-04-14:
The issue is that PD8/PD2 do not have alternate functions for UART2.  And PD0/PD1 do not have alternate functions for UART4.  Same for the UART5 changes.  You must select pins that support the alternate function (only certain pins support certain UARTs).

closed 2021-04-14 by dpgeorge