micropython/micropython #6592: .bss corruption with native modules on ESP32

jimmo, 2020-11-02:
Modify features1.c to change `uint16_t data16[4];` to `uint16_t data16[32];`.
Compile for xtensawin
Copy over features1.mpy and test.py containing:

```py
import gc
import features1

def demo():
    print(features1.access())
    features1.access()
    features1.access()
    gc.collect()

demo()
```

Import test then run test.demo() a few times. The list printed out by `features1.access()` should be all-zero but contains modified entries.

```
>>> import test
initialising module self=3ffe5440
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
>>> test.demo()
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
>>> test.demo()
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
>>> test.demo()
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 3, 0, 1, 0, 3, 0, 1, 0, 7, 0, 1, 0, 3, 0, 1, 0]
>>> test.demo()
[7, 0, 1, 0, 3, 0, 1, 0, 15, 0, 1, 0, 3, 0, 1, 0, 7, 0, 1, 0, 3, 0, 1, 0, 31, 0, 1, 0, 3, 0, 1, 0]
>>> test.demo()
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 7, 0, 1, 0, 3, 0, 1, 0, 31, 0, 1, 0, 3, 0, 1, 0]
>>> test.demo()
[7, 0, 1, 0, 3, 0, 1, 0, 15, 0, 1, 0, 3, 0, 1, 0, 7, 0, 1, 0, 3, 0, 1, 0, 31, 0, 1, 0, 3, 0, 1, 0]
```

This happens on both spiram and non-spiram builds. Making `data16` larger (or smaller) seems to make the issue go away?

jimmo, 2020-11-02:
Originally reported at the forum: https://forum.micropython.org/viewtopic.php?f=3&t=9243&p=52062

dpgeorge, 2020-11-05:
Thanks for reporting.  This is most likely the same issue as #6045 : the top-level function object for a module (which executes the module's outer scope) usually has nothing pointing to it after it has executed, so can be GC'd.  For native .mpy files this top-level function object points to the bss and rodata (if they exist), so they'll get GC'd.  And this problem should manifest on all architectures.

The fix can be the same as in #6045, to maintain a list of all imported native .mpy top-level function objects, but it'd be nice to find a neater way to do it....

agners, 2020-11-15:
@dpgeorge sounds plausible. I tried the the suggested change in https://github.com/micropython/micropython/pull/6061/files, which from what I understand does keep a global pointer for __init__ (and with that bss). With this change I can no longer reproduce the issue...

agners, 2020-12-26:
Unfortunately, it seems not to do the trick entirely: After some hours/days of running, I get `FAT filesystem appears to be corrupted.`...

PR #15908 (2024-09-26): py: Properly track root pointers for text, BSS and rodata imported from native .mpy files
### Summary

When importing a `.mpy` file with native code, memory is allocated to hold the text of the native code (the machine code), and also optionally a BSS/rodata section.

On certain architectures, the text may be allocated in a special region, outside the MicroPython GC heap.  On other architectures it's allocated on the GC heap.

The BSS/rodata is always allocated on the GC heap.

The trick is making sure pointers to GC-allocated memory are retained and reachable by the GC.  This is difficult bec [... cut]
closed 2024-09-26 by dpgeorge