micropython/micropython #6599: EM_X86_64 with R_386_GOTOFF is not supported in text relocation

afakhry01, 2020-11-04:
I get an assertion error when trying to build an mpy module from c code. I am using the code provided in the docs (https://docs.micropython.org/en/latest/develop/natmod.html). The error is as follow:
```
Traceback (most recent call last):
  File "../../build/micropython//tools/mpy_ld.py", line 1082, in <module>
    main()
  File "../../build/micropython//tools/mpy_ld.py", line 1078, in main
    do_link(args)
  File "../../build/micropython//tools/mpy_ld.py", line 1049, in do_link
    link_objects(env, len(native_qstr_vals), len(native_qstr_objs))
  File "../../build/micropython//tools/mpy_ld.py", line 831, in link_objects
    do_relocation_text(env, sec.addr, r)
  File "../../build/micropython//tools/mpy_ld.py", line 577, in do_relocation_text
    assert 0, r_info_type
AssertionError: 9
```
And those are my settings:
```
env.arch.name is EM_X86_64
r_info_type is 9 (R_386_GOTOFF)
```

I am not sure how these GOT calculations are made. I might be able to work on it if any instructions are provided. Thanks a lot!

dpgeorge, 2020-11-04:
Thanks for the report.  I'm not sure how it's getting an `R_386_GOTOFF` relocation on an x86-64... can you please provide full information about your build system: OS and C compiler.  Also it would help to dump the .o file that is generated by the C compiler using `objdump -xD build/test.o`, for example.

afakhry01, 2020-11-04:
Thanks for replying. My operating system is `SUSE Linux` and I am using `gcc 4.8.5`. I was also running `mpy_ld.py` from a `conda` environment. My `objdump` is:
```
factorial.o:     file format elf64-x86-64
factorial.o
architecture: i386:x86-64, flags 0x00000011:
HAS_RELOC, HAS_SYMS
start address 0x0000000000000000

Sections:
Idx Name          Size      VMA               LMA               File off  Algn
  0 .text         00000068  0000000000000000  0000000000000000  00000040  2**0
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE
  1 .data         00000000  0000000000000000  0000000000000000  000000a8  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00000000  0000000000000000  0000000000000000  000000a8  2**0
                  ALLOC
  3 .data.rel.ro  00000010  0000000000000000  0000000000000000  000000b0  2**4
                  CONTENTS, ALLOC, LOAD, RELOC, DATA
  4 .comment      00000019  0000000000000000  0000000000000000  000000c0  2**0
                  CONTENTS, READONLY
  5 .note.GNU-stack 00000000  0000000000000000  0000000000000000  000000d9  2**0
                  CONTENTS, READONLY
  6 .eh_frame     00000060  0000000000000000  0000000000000000  000000e0  2**3
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, DATA
SYMBOL TABLE:
0000000000000000 l    df *ABS*	0000000000000000 factorial.c
0000000000000000 l    d  .text	0000000000000000 .text
0000000000000000 l    d  .data	0000000000000000 .data
0000000000000000 l    d  .bss	0000000000000000 .bss
0000000000000000 l     F .text	000000000000002f factorial
0000000000000000 l     O .data.rel.ro	0000000000000010 factorial_obj
0000000000000000 l    d  .data.rel.ro	0000000000000000 .data.rel.ro
0000000000000000 l    d  .note.GNU-stack	0000000000000000 .note.GNU-stack
0000000000000000 l    d  .eh_frame	0000000000000000 .eh_frame
0000000000000000 l    d  .comment	0000000000000000 .comment
0000000000000000         *UND*	0000000000000000 _GLOBAL_OFFSET_TABLE_
0000000000000000         *UND*	0000000000000000 mp_fun_table
000000000000002f g     F .text	0000000000000039 mpy_init
0000000000000000         *UND*	0000000000000000 mp_native_qstr_val_table
0000000000000000         *UND*	0000000000000000 mp_type_fun_builtin_1

Disassembly of section .text:

0000000000000000 <factorial>:
   0:	53                   	push   %rbx
   1:	48 8b 1d 00 00 00 00 	mov    0x0(%rip),%rbx        # 8 <factorial+0x8>
			4: R_X86_64_GOTPCREL	mp_fun_table-0x4
   8:	be 02 00 00 00       	mov    $0x2,%esi
   d:	ff 53 18             	callq  *0x18(%rbx)
  10:	bf 01 00 00 00       	mov    $0x1,%edi
  15:	48 85 c0             	test   %rax,%rax
  18:	74 09                	je     23 <factorial+0x23>
  1a:	48 0f af f8          	imul   %rax,%rdi
  1e:	48 ff c8             	dec    %rax
  21:	eb f2                	jmp    15 <factorial+0x15>
  23:	48 8b 43 20          	mov    0x20(%rbx),%rax
  27:	be 02 00 00 00       	mov    $0x2,%esi
  2c:	5b                   	pop    %rbx
  2d:	ff e0                	jmpq   *%rax

000000000000002f <mpy_init>:
  2f:	41 54                	push   %r12
  31:	55                   	push   %rbp
  32:	53                   	push   %rbx
  33:	48 8b 1d 00 00 00 00 	mov    0x0(%rip),%rbx        # 3a <mpy_init+0xb>
			36: R_X86_64_GOTPCREL	mp_fun_table-0x4
  3a:	48 8b 7f 08          	mov    0x8(%rdi),%rdi
  3e:	48 8b 6b 28          	mov    0x28(%rbx),%rbp
  42:	ff d5                	callq  *%rbp
  44:	49 89 c4             	mov    %rax,%r12
[... 55 lines cut ...]
  24:	2f                   	(bad)  
  25:	00 00                	add    %al,(%rax)
  27:	00 00                	add    %al,(%rax)
  29:	41 0e                	rex.B (bad) 
  2b:	10 83 02 6c 0e 08    	adc    %al,0x80e6c02(%rbx)
  31:	00 00                	add    %al,(%rax)
  33:	00 28                	add    %ch,(%rax)
  35:	00 00                	add    %al,(%rax)
  37:	00 38                	add    %bh,(%rax)
  39:	00 00                	add    %al,(%rax)
  3b:	00 00                	add    %al,(%rax)
			3c: R_X86_64_PC32	.text+0x2f
  3d:	00 00                	add    %al,(%rax)
  3f:	00 39                	add    %bh,(%rcx)
  41:	00 00                	add    %al,(%rax)
  43:	00 00                	add    %al,(%rax)
  45:	42 0e                	rex.X (bad) 
  47:	10 8c 02 41 0e 18 86 	adc    %cl,-0x79e7f1bf(%rdx,%rax,1)
  4e:	03 41 0e             	add    0xe(%rcx),%eax
  51:	20 83 04 71 0e 18    	and    %al,0x180e7104(%rbx)
  57:	41 0e                	rex.B (bad) 
  59:	10 42 0e             	adc    %al,0xe(%rdx)
  5c:	08 00                	or     %al,(%rax)
	...
```

dpgeorge, 2020-11-05:
Thanks for the output.  It looks like that version of gcc is generating `R_X86_64_GOTPCREL` relocations instead of `R_X86_64_REX_GOTPCRELX` (which I see locally with gcc v10).

Looking at https://reviews.llvm.org/D18301 it seems that these can be treated equally in the linker.

So the following patch may fix it:
```diff
diff --git a/tools/mpy_ld.py b/tools/mpy_ld.py
index 774966a7f..6df61bb2f 100755
--- a/tools/mpy_ld.py
+++ b/tools/mpy_ld.py
@@ -73,6 +73,7 @@ R_XTENSA_SLOT0_OP = 20
 R_ARM_BASE_PREL = 25  # aka R_ARM_GOTPC
 R_ARM_GOT_BREL = 26  # aka R_ARM_GOT32
 R_ARM_THM_JUMP24 = 30
+R_X86_64_GOTPCREL = 9
 R_X86_64_REX_GOTPCRELX = 42
 R_386_GOT32X = 43
 
@@ -132,7 +133,7 @@ ARCH_DATA = {
         | MICROPY_OPT_CACHE_MAP_LOOKUP_IN_BYTECODE,
         2,
         8,
-        (R_X86_64_REX_GOTPCRELX,),
+        (R_X86_64_GOTPCREL, R_X86_64_REX_GOTPCRELX),
         asm_jump_x86,
     ),
     "armv7m": ArchData(
@@ -536,7 +537,7 @@ def do_relocation_text(env, text_addr, r):
         # Relcation pointing to GOT
         reloc = addr = env.got_entries[s.name].offset
 
-    elif env.arch.name == "EM_X86_64" and r_info_type == R_X86_64_REX_GOTPCRELX:
+    elif env.arch.name == "EM_X86_64" and r_info_type in (R_X86_64_GOTPCREL, R_X86_64_REX_GOTPCRELX):
         # Relcation pointing to GOT
         got_entry = env.got_entries[s.name]
         addr = env.got_section.addr + got_entry.offset
```

afakhry01, 2020-11-05:
I just wanted to confirm that the issue was fixed with the given patch. Thanks a lot!

PR #6619 (2021-05-20): tools/mpy_ld.py: Support R_X86_64_GOTPCREL reloc for x86-64 arch.
This can be treated by the linker the same as R_X86_64_REX_GOTPCRELX, according to https://reviews.llvm.org/D18301.

Fixes #6599
closed 2021-05-20 by dpgeorge