micropython/micropython #6601: BLE events getting "stuck" on ESP32 devices

jonathanhogg, 2020-11-05:
I'm writing code for multiple ESP32 devices that communicate with each other over BLE, acting as both central and peripheral to each other in a kind of simple mesh network over the Nordic BLE UART protocol. The code makes heavy use of `uasyncio`, with everything running as tasks. I sync the BLE IRQ callback to the task that handles events through the simple mechanism of pumping the events into a list queue (copying all of the arguments) and then having the Bluetooth task wake every few milliseconds and check this list.

Mostly this is working completely fine, but I'm seeing regular situations where I start a connection to a peripheral and go back to the main loop, which can run away happily for several minutes without any sign that the connection has gone anywhere. If I interrupt execution and return to the prompt I'll *then* see a `PERIPHERAL_CONNECT`, subsequent `PERIPHERAL_DISCONNECT` and maybe a few other BLE events depending on what else was going on – like a `CENTRAL_CONNECT` and `CENTRAL_DISCONNECT` from another one of my units trying to query this one.

I've seen some worrying delays occasionally in central connections and GATT write/notify operations, but nothing of the same magnitude as these peripheral connections that just disappear completely.

With some additional debug prints, I'm certain that the error is my `BLE.irq()` registered method not being called rather than any problem with the transfer of events from callback land to async world. I previously used a more complicated mechanism for doing this using a custom IRQ-safe stream reader / ioctl, but I was seeing the same errors there and so I went with the simpler approach.

Trawling through the code, the only way I can see that this could be occurring is if the scheduler has stopped processing pending callbacks – which seems unlikely since calls to `mp_handle_pending()` seem to be everywhere in the MicroPython code – or the Nimble stack has stopped reporting them. The fact that the events appear immediately on interrupting my code is super-suspicious however.

Next step is that I'm going to put something in to keep scheduling a dummy callback so that I can see if the scheduler is doing its work.

I'm running my code against a fairly recent mainline build with only the addition of a peripheral connection cancellation method (as per PR #6584), which I added when I thought that the problem was network timeouts.

jonathanhogg, 2020-11-05:
OK, so it's definitely not the scheduler: I added a task that schedules a callback every 100ms and then checks that this has been invoked. No problem here, my callbacks are being called basically as soon as they've been scheduled – presumably the VM is doing it at the next instruction boundary.

However, here's a log excerpt from my app (including this check running throughout):

```
D (3694.544) bluart: Begin peripheral connection to Address(<public:3c71bf87fd1a>)
D (3698.063) rainbow: Frame rate: 98
D (3708.143) rainbow: Frame rate: 99
D (3718.234) rainbow: Frame rate: 99
D (3724.638) bluart: Cancel peripheral connection attempt
D (3728.508) rainbow: Frame rate: 97
D (3738.598) rainbow: Frame rate: 99
D (3748.678) rainbow: Frame rate: 99
E (3754.663) mesh: Unexpected exception
Traceback (most recent call last):
  File "mesh.py", line 172, in peripheral_connect
  File "bluart.py", line 394, in connect
ConnectionError: Peripheral connection attempt is stuck
MicroPython v1.12-854-g03f16151a-dirty on 2020-11-05; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 3754703 7 (1, 0, <memoryview>)
3754704 1 (0, 0, <memoryview>)
3754712 21 (0, 515)
3754714 3 (0, 21)
3754714 8 (1, 0, <memoryview>)
3754722 2 (0, 0, <memoryview>)
```

Log line timestamps are in seconds. The `Frame rate` lines are being spat out by the main task every few seconds to show it's running normally. 30 seconds after calling `gap_connect()`, I have a go at cancelling it (using new mechanism from #6584) and then wait another 30 seconds before just exploding the `uasyncio` loop.

After the return to the prompt you see 6 calls to the IRQ callback, which has a `print()` stuck in it outputting a timestamp in milliseconds, event number and event arguments. The events are: the missing `PERIPHERAL_CONNECT`; a `CENTRAL_CONNECT` from another device making an attempt to talk to this one; an `MTU_EXCHANGED` for the central connection; a `GATTS_WRITE` from the central, a `PERIPHERAL_DISCONNECT` for the timed-out peripheral connection attempt; and a `CENTRAL_DISCONNECT` from the timed-out central. So lots has been going on in the meantime.

I was watching this happen and all 6 of these callbacks occurred immediately on return to the prompt, so they appear to have been queued somewhere inside the Nimble stack, not in MicroPython. I'm not sure what it is about the return to the prompt that is triggering the delivery of them. I haven't delved into the Nimble code yet, is it a thread reading events from the underlying hardware that has become stuck? A race condition on a lock?

jonathanhogg, 2020-11-05:
I have three more examples of this failure now and they all involve a background central connection coming in just after (judging by the event ordering) the outgoing peripheral connection.

So it seems plausible that there is a problem somewhere in the stack dealing with this scenario.

jonathanhogg, 2020-11-05:
Rats! It's not that. I had an idea for a workaround: if it's clash between incoming and outgoing connections, then advertise a couple of seconds before making an outgoing connection that we're no longer accepting connections. Peer devices see the advert and leave this device alone. This works, but I still see stuck events:

```
D (16510.374) rainbow: Frame rate: 92
D (16517.751) bluart: Disallow connections
D (16519.792) bluart: Begin peripheral connection to Address(<public:3c71bf87fd1a>)
16519813 6 ()
D (16519.850) bluart: Scanning stopped
D (16520.839) rainbow: Frame rate: 95
D (16530.547) rainbow: Frame rate: 103
D (16540.130) rainbow: Frame rate: 104
D (16549.719) rainbow: Frame rate: 104
D (16549.868) bluart: Cancel peripheral connection attempt
D (16559.186) rainbow: Frame rate: 105
D (16568.713) rainbow: Frame rate: 104
D (16578.138) rainbow: Frame rate: 106
E (16579.890) mesh: Unexpected exception
Traceback (most recent call last):
  File "mesh.py", line 180, in peripheral_connect
  File "bluart.py", line 394, in connect
ConnectionError: Peripheral connection attempt is stuck
MicroPython v1.12-854-g03f16151a-dirty on 2020-11-05; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 16579928 7 (0, 0, <memoryview>)
16579937 8 (0, 0, <memoryview>)
```

No background central connection this time, just `CONNECTED` and `DISCONNECTED` events showing up a minute after calling `gap_connect()` and only when the main loop has been interrupted.

tve, 2020-11-05:
Looking at your logs... you're not just getting a ctrl-C, you're getting a soft reset. If it were just a ctrl-C you would get a `>` prompt after the `ConnectionError` line. But you're getting the `MicroPython v1.12-...` prompt. If you look into the esp32 `main.c`  the first thing executed is `mp_bluetooth_deinit();` etc. Hope this helps...

jonathanhogg, 2020-11-05:
That would likely be because the code is executing directly from `boot.py`. So when it exits, it falls out of that into the command line interpreter.

However, this is a good spot, because it may explain why the events are suddenly becoming "unstuck" from wherever they were. I'll need to stick my head a bit more into the MicroPython source to trace exactly what happens…

jimmo, 2020-11-06:
@jonathanhogg I will have to spend some more time thinking about this, but there is one thing that would definitely explain missing events is when the ringbuffer becomes full. So in a situation with many events arriving concurrently this is definitely possible. Have you tried experimenting with `ble.config(rxbuf=N)` where N is something much larger than the default of 128.

Are you possibly able to share the code so I can try and repro the issue here? Privately is fine (either private github repo or by email or something).

FWIW in #6594 I'm trying to get rid of the ringbuffer, but doing this for ESP32 will be a future PR (see notes in the PR description).

jonathanhogg, 2020-11-06:
@jimmo, I actually set `rxbuf` to 1K before doing anything else. I'd be surprised if it was that as – from my reading of the code – I would expect any queued events to be getting pushed out of it during the period where the connection seems to be stuck.

My code is fairly sprawling I'm afraid, but you are welcome to try it out. I'll set you up with access to our private Gitea repo. I'm on the clock now for figuring out a fix or a workaround as this code is going into a pair of massive light artworks shortly…

jonathanhogg, 2020-11-14:
I can't really make a lot of sense of this, but I've got the code to be pretty much stable if I start advertising that a node is unable to accept connections 5 seconds before attempting an outgoing connection. Weirdly this makes a difference regardless of whether or not any other device attempts to connect to it in this period.

If I don't do this then I see multiple problems with events getting stuck – most often notifications from the peripheral that seem to have disappeared into the aether, only to suddenly appear the moment my code times-out and drops the connection.

tve, 2020-12-23:
FYI, your woes are most likely due to the fact that uasyncio is incompatible with events coming in via ISRs, including soft ISRs (mp_sched_schedule). If you're willing to compile micropython yourself you can add a very simple hack. First add a ticker task that just calls `asyncio.sleep(1000)` is a loop, this prevents asyncio from exiting when all tasks are waiting for events. Then make the following change:
```
diff --git a/extmod/uasyncio/core.py b/extmod/uasyncio/core.py
index 6a84b0982..667b86c29 100644
--- a/extmod/uasyncio/core.py
+++ b/extmod/uasyncio/core.py
@@ -164,6 +164,8 @@ def run_until_complete(main_task=None):
                 # No tasks can be woken so finished running
                 return
             # print('(poll {})'.format(dt), len(_io_queue.map))
+            if dt > 5:
+                dt = 5
             _io_queue.wait_io_event(dt)

         # Get next task to run and continue it
```
This causes asyncio never to sleep more than 5ms and means that if an asyncio task gets unblocked via a bluetooth callback it gets picked up within 5ms.

jonathanhogg, 2020-12-24:
Thanks @tve, but I'm already using a custom `IOBase` class to adapt from MicroPython scheduler-land to `uasyncio`-land (following the examples from @peterhinch) and the rest of my async code is running fine, processing hundreds of context switches per second across several tasks.

Instrumentation has shown that the issue is a failure to schedule the BLE `.irq()` callback itself. I've got the code stable at the moment (it's already been baked into 46 controllers running three live public artworks), largely through reducing BLE activity. So I'm leaning towards thinking that it's some kind of deadlock inside the stack. I've not tried the new ring-buffer replacement code yet to see if that makes any kind of difference.

@jimmo, I also found that receiving several – 4 or 5 – back-to-back notify events containing data (in total much smaller than my 2KB ring-buffer size) would lock up the callback mechanism. The code would receive only the first 1 or 2 events, then time-out the connection after 30 seconds and close it, which would immediately trigger the missing notify callbacks followed by the disconnect event. I worked around this by ensuring that my protocol sticks to strict interleaved unconfirmed-write/notify-with-data messaging between peers.

If I get some free time early next year I'll try to reduce the code down to a small failing example and bury some instrumentation deeper into the stack.

pumelo, 2020-12-25:
Without knowing anything about BLE: I had strange issues on the esp32 due to freertos task priority inversion.

dpgeorge, 2021-04-14:
This issue will hopefully be made obsolete (ie fixed) by #7046.

dpgeorge, 2022-07-22:
BLE on esp32 now uses synchronous events (see 5dbb822ca4a809ac5cb4513afb0411b4eb8dc3cf), so hopefully that fixes this issue.  Feel free to reopen if not.

closed 2022-07-22 by dpgeorge