micropython/micropython #6604: Cannot access webrepl on Pyboard D if main thread is running

Dr-StephenStrange, 2020-11-06:
While the main thread is running e.g.

`While True:
    Do Stuff
    sleep(1)
`
I cannot access the webrepl prompt >>>

I've tried all different types of sleep e.g. ms, us

Until the main thread finishes there's no way to get a webrepl working.

This limits the use of the Pyboard D for remote systems, e.g. Data Loggers outside

kevinkk525, 2021-05-23:
Glad I found this issue, so it seems it has been ignored since over half a year now. Guess my issue #7208 is a duplicate.
So it seems the expensive pyboard-D is even less worth than an esp8266 in my use-case. What a shame.

Matrosskin, 2022-11-01:
I have same problem with esp32: I can't access webrepl while the loop is in progress.