micropython/micropython #6608: `float` representation of objects with buffers

v923z, 2020-11-09:
This issue came up as a side comment in https://github.com/v923z/micropython-ulab/issues/208, and in short, the following produces a nasty segmentation fault: 

```
>>> a = np.array([[1,2,3],[4,5,6],[7,8,9]], dtype=np.int8)
>>> float(a)
Segmentation fault (core dumped)
```
I believe, the problem is the default case of `float_make_new`: https://github.com/micropython/micropython/blob/b7883ce74c5a9b9689d812d134117d625fd42e73/py/objfloat.c#L128-L133

The `ndarray` does have a buffer, but the `mp_parse_num_decimal` function is supposed to work with strings only. I have the feeling that any user type that defines a buffer would suffer from this issue, and this is why I would like to suggest that it be fixed here, and not in `ulab`. I could take a stab at providing a fix. I assume, one only has to check, whether the argument is of type string. Is that correct?

peterhinch, 2020-11-10:
> I have the feeling that any user type that defines a buffer would suffer from this issue

Can you demonstrate an instance where this actually occurs using standard Python types rather than `ulab` types?

I passed various types to the **float()** method and in every case a **ValueError** was the outcome rather than a crash. For example **bytearray** and integer **array** instances behave properly.

v923z, 2020-11-10:
> > I have the feeling that any user type that defines a buffer would suffer from this issue
> 
> Can you demonstrate an instance where this actually occurs using standard Python types rather than `ulab` types?

If we stay within the bounds of `micropython`, then this will most probably not happen. But if you look at the snippet above, the crash is caused by `mp_parse_num_decimal(bufinfo.buf, bufinfo.len, false, false, NULL); `, https://github.com/micropython/micropython/blob/073b9a5eb888b26be7ce4c65e9649b6c3857ba39/py/parsenum.c, which seems to go through various cases, but they are all predictable within `micropython`.

> I passed various types to the **float()** method and in every case a **ValueError** was the outcome rather than a crash. For example **bytearray** and integer **array** instances behave properly.

`ValueError` is indeed the very last case in https://github.com/micropython/micropython/blob/073b9a5eb888b26be7ce4c65e9649b6c3857ba39/py/parsenum.c#L354

But your comment seems to indicate that I could guard against this error on `ulab`'s side. Do you know what is required for that? I don't immediately see, where I could hook into the function.

peterhinch, 2020-11-12:
I was puzzled whether you were reporting a bug or requesting a new feature. To the Python programmer it seems correct. In every case I tested, including custom classes, behaviour is as per CPython.

You evidently need an interface (exception?) at the C level. Alas this is rather above my pay grade; I'll leave the issue to you experts.

v923z, 2020-11-12:
> I was puzzled whether you were reporting a bug or requesting a new feature. 

Sorry, then I didn't properly articulate the problem. It is a bug.

> To the Python programmer it seems correct. In every case I tested, including custom classes, behaviour is as per CPython.

Do you mean `python` classes, or classes implemented in C?

peterhinch, 2020-11-13:
Python. From a Python perspective everything seems OK.

dpgeorge, 2021-04-14:
The code you posted at the top should not crash like that, even if  `ndarray` does have a buffer.

Can you provide a minimal example in C (eg add some code to `ports/unix/coverage.c`) which shows the crash?

v923z, 2021-04-14:
> The code you posted at the top should not crash like that, even if `ndarray` does have a buffer.

With the most recent version, I only get a `ValueError`, and `micropython` doesn't crash anymore. I presume, this was fixed sometime between now and last November. 
```
from ulab import numpy

>>> a = numpy.array([[1, 2, 3], [4, 5, 6], [7, 8, 9]], dtype=numpy.uint8)
>>> a
array([[1, 2, 3],
       [4, 5, 6],
       [7, 8, 9]], dtype=uint8)
>>> float(a)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: invalid syntax for number
```
I think this can be closed.

closed 2021-04-14 by v923z

dpgeorge, 2021-04-15:
Thanks for following up.  Feel free to reopen the issue if the problem is seen again.