micropython/micropython #6618: ESP32: dhcp_hostname is read-only on non-WLAN interfaces

scy, 2020-11-13:
Hi there!

On a `network.WLAN` interface, I can set `.config(dhcp_hostname='galactica')`. However, on a LAN interface, I cannot:

```
>>> import network
>>> l = network.LAN()
>>> l.config('dhcp_hostname')
'espressif'
>>> l.config(dhcp_hostname='galactica')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: unknown config param
```

I was trying to do this on an Olimex ESP32-POE board, i.e. `phy_type=network.PHY_LAN8720`, but as far as I understand, the exact interface doesn’t matter since this config option will simply be passed to the ESP32’s DHCP client.

Would be cool if I could set a hostname also for a wired board! Thanks in advance for looking into this.

lovetheguitar, 2021-04-26:
Yes please, we experience the same issue. 

As a side note, anything related to mDNS does not seem to be supported on wired connections as well sadly. 

Thanks a lot.

Thijxx, 2022-09-12:
Same issue on network.WLAN

```
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    wlan.config(hostname="himom")
    wlan.connect(ssid, password)
```