micropython/micropython #6621: Uaysncio v3 Stream Writer Drain Bug

gtomassi, 2020-11-14:
The new implementation of Stream will write redundant frames out it's socket/serial connection if used asynchronously.

**Expected Behavior:**
This pseudo-code should write N packets out the uart,
```
streamWriter = Stream(uart...
async def foo():
          streamWriter.write(packet)
          await streamWriter.drain()
for i in range(0,N):
          asyncio.create_task(foo())
```
but instead will write the first packet N times, the second packet N-1 times, ... the last packet one time.

gtomassi, 2020-11-14:
@dpgeorge @peterhinch 
This may be one to prioritize as it does immediately exhibit symptoms for anyone upgrading to v3. In particular it will likely happen when people queue up >1 of the legacy awrite's in succession.

If there is anything we can do to test a fix, we are happy to please let me know. Thank you.

peterhinch, 2020-11-15:
This is not a safe way to use a StreamWriter.

At a hardware level a stream device may buffer **C** characters before temporarily stalling while these are physically transmitted. Assume **C** << **len(packet)**. The currently transmitting task sends **C** characters, the hardware signals "full" via the ioctl and the task yields to the scheduler. When the hardware has space in its buffer the ioctl allows one of the tasks waiting on the StreamWriter to send a batch of characters. It won't be the task which sent the last batch.

The result will be garbled messages.

The correct way to use StreamWriter with multiple tasks is to implement a queue. Multiple tasks put messages on the queue, and a single task removes them and issues them to the **StreamWriter**. No other task accesses the **StreamWriter**.

Note that this logic applies to most approaches to asynchronous handling of streams and not tied to the specific implementation of uasyncio.

jonnor, 2024-09-13:
This seems to have been answered.

closed 2024-09-13 by jonnor