micropython/micropython #6623: machine.I2C.init will fail on hardware I2C

jimmo, 2020-11-15:
Reported on the forum here: https://forum.micropython.org/viewtopic.php?f=2&t=9300&p=52329#p52309

This is ESP32, but I think this will apply on other ports too. The port-specific hardware I2C type (now machine.I2C) uses the software I2C's locals dict, so init() will try and initialise assuming it's a machine.SoftI2C instance.

example code:

```py
i = machine.I2C(id, scl=, sda=)  # Hardware machine.I2C
i.init(...) # calls machine_i2c_obj_init_helper, assuming self is a mp_machine_soft_i2c_obj_t
```

peterhinch, 2020-11-15:
I have been getting strange results from ESP32 I2C, but haven't yet fully characterised what's going on. The following works:
```python
i2c = I2C(1, scl=Pin(32), sda=Pin(33), freq=1_000_000)
```
but the following, taken from the docs, does not:
```python
i2c = I2C(0)  # Expected scl==18, sda==19
```
I got timeouts from I2C(0) even if I explicitly specified pins 18 and 19.

dpgeorge, 2020-11-18:
> This is ESP32, but I think this will apply on other ports too. The port-specific hardware I2C type (now machine.I2C) uses the software I2C's locals dict, so init() will try and initialise assuming it's a machine.SoftI2C instance.

Thanks for reporting.  Looks like this has been an issue for a while (prior to I2C/SPI changes), I confirmed that the bug is in esp32 and stm32 at v1.13.

> The following, taken from the docs, produces ENODEV:
> 
> ```python
> i2c = I2C(0)  # Hardware on scl==18, sda==19
> ```

I can't reproduce this, it constructs the I2C(0) OK for me:
```
MicroPython v1.13-173-g61d1e4b01 on 2020-11-18; TinyPICO with ESP32-PICO-D4
Type "help()" for more information.
>>> import machine
>>> machine.I2C(0)
I2C(0, scl=18, sda=19, freq=400000)
>>> machine.I2C(1)
I2C(1, scl=25, sda=26, freq=400000)
>>> machine.I2C(2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: I2C(2) doesn't exist
>>> 
```

dpgeorge, 2020-11-18:
The issue with init crashing should be fixed by #6630 (it now raises an exception calling `.init()` on HW I2C, it can be implemented in the future if/when needed).

peterhinch, 2020-11-18:
> I can't reproduce this, it constructs the I2C(0) OK for me:

Sorry, I evidently did a poor job of explaining this. It does indeed construct OK. The interface produces `ENODEV` when you try to use it with hardware connected to those pins. Tested with the official LCD160CR board and the ESP32 reference board. (I did have pullups).

It works when the pins are explicitly specified to the constructor.

I guess the fault is in the docs.

peterhinch, 2020-11-20:
I have raised https://github.com/micropython/micropython/issues/6637 in an attempt to clarify this.

dpgeorge, 2020-11-23:
The issue of `.init()` crashing on HW I2C is fixed by 64180f0742a1926162b784c68bafa49b9b58596c

closed 2020-11-23 by dpgeorge