micropython/micropython #6635: real and imag raise exceptions on non-complex numbers

HugoNumworks, 2020-11-19:
The real or imaginary part of a non-complex number cannot be computed using MicroPython, whereas Python and CPython handle it.
Should I make a PR for this ?
It would be handled with MICROPY_PY_BUILTINS_COMPLEX only.
[image]

dlech, 2020-11-19:
What is the use case for this?

Is `complex(2).real` a suitable workaround?

peterhinch, 2020-11-19:
I guess the use case is where at runtime a variable might contain a real or a complex.

HugoNumworks, 2020-11-20:
Indeed, but it would have been preferable not having to consider the number's type, like with Python. Same might go with conjugate() method.