micropython/micropython #6636: Building Unix port on Raspberry Pi 4 flags compilation error!

mouryarahul, 2020-11-19:
Hi, I'm trying to build Unix port of Micropython on Raspberry Pi 4 (Ubuntu 20.10):
`Linux rahul-pi4 5.8.0-1007-raspi #10-Ubuntu SMP PREEMPT Thu Nov 5 17:52:40 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux`. Building `mpy-cross` was successful, but while compiling `unix port`, I'm getting following error:
`Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
CC build-standard/frozen_content.c
CC main.c
main.c: In function ‘main_’:
main.c:595:22: error: variable ‘subpkg_tried’ might be clobbered by ‘longjmp’ or ‘vfork’ [-Werror=clobbered]
  595 |                 bool subpkg_tried = false;
      |                      ^~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:77: build-standard/main.o] Error 1
`
What will be the workaround for this error? Thanks for your suggestions.

mouryarahul, 2020-11-19:
> Hi, I'm trying to build Unix port of Micropython on Raspberry Pi 4 (Ubuntu 20.10):
> `Linux rahul-pi4 5.8.0-1007-raspi #10-Ubuntu SMP PREEMPT Thu Nov 5 17:52:40 UTC 2020 aarch64 aarch64 aarch64 [...]

Update: I'm able to successfully build STM32 port.

mouryarahul, 2020-11-19:
> Update: I'm able to successfully build STM32 port.

Update: I added `volatile` keyword for the variable:
`bool subpkg_tried = false;` and the compilation error is suppressed. I guess this something to do with the way this variable is being used!

dpgeorge, 2020-11-20:
> Update: I added `volatile` keyword for the variable

Yes that's probably the correct solution here.  I noticed this same warning on the esp32 port with the latest toolchain, so that's another place this fix can be tested before committing it to master.

mouryarahul, 2020-11-20:
@dpgeorge I'm not sure what changed, but when I checked out `v1.13`, the compilation problem disappeared without any modification in the source code.

embeddedt, 2021-03-14:
It looks like this issue is still happening on 1.14, but only with certain compilers. Compiling on an x64 desktop doesn't seem to trigger it. However, an LVGL user reported getting this error with an Armv8 compiler. https://forum.lvgl.io/t/invalid-arch-error/4898/5?u=embeddedt

I suspect that it only appears on certain platforms now because the optimizer hides the problem.

dpgeorge, 2021-04-14:
Should be fixed by 9c9bfe1968d4260924316353bd1c941fe7e8741d

closed 2021-04-14 by dpgeorge