micropython/micropython #6637: ESP32: Hardware I2C constructor does not conform with docs

peterhinch, 2020-11-20:
In https://github.com/micropython/micropython/issues/6623 I evidently didn't explain the problem well so I've raised a new issue.

[The docs](http://docs.micropython.org/en/latest/esp32/quickref.html#hardware-i2c-bus) state that it is possible to instantiate the two hardware I2C buses on default pins as follows:
```python
i2c0 = I2C(0)  # Pins 18 (scl) and 19 (sda)
i2c1 = I2C(1)  # Pins 25 and 26
```
Construction occurs without error, but the interface is unusable on those pins, throwing an ENODEV on first usage. The interface works correctly on these pins if the pins are explicitly specified:
```python
i2c = I2C(0, scl=Pin(18), sda=Pin(19), freq=1_000_000)
```
Tested on the reference board connected to the official LCD160CR display. With pullups :)

I am unsure whether the fault lies in the code, the docs, or the default pin numbers.

dpgeorge, 2020-11-23:
I'm still not able to reproduce this... I have tried the following (I'm using a TinyPICO for the esp32 board which is all I have on me at the moment):
1. Plug a TinyPICO and PYBv1.0 into the same PC via their USB ports.
2. Connect TinyPICO pins 18/19 to PYBv1.1 pins X9/X10 (as well as GND-to-GND).
3. Run `pyb.Accel()` on the pyboard to power up the accelerometer I2C device.
4. Then run and get the following on the TinyPICO:
```
MicroPython v1.13-181-g85ddc445e on 2020-11-18; TinyPICO with ESP32-PICO-D4
Type "help()" for more information.
>>> import machine
>>> i2c0=machine.I2C(0)
>>> i2c1=machine.I2C(1)
>>> i2c0.scan()
[76]
>>> i2c0.readfrom_mem(76, 0, 8)
b'\x10;3\x16\x01\x00\x00\x01'
>>> 
```

peterhinch, 2020-11-24:
I have reproduced your results on the reference board.

I evidently diagnosed this problem incorrectly. I'll investigate further and (if appropriate) raise a new issue.

closed 2020-11-24 by peterhinch

dpgeorge, 2020-11-24:
Sounds good!

peterhinch, 2020-11-30:
I believe - but haven't proved - this is a timing problem in the LCD160CR firmware. I was attempting to access I2C before the device was ready. Now fixed by a workround in my code.

In interfacing the official LCD160CR to an ESP32 I encountered a few points which might influence a new hardware/firmware release. I'll raise an RFC so that there is a record of these.