micropython/micropython #6641: Simultaneous LAN8720 and WIFI connection will render ESP32-WROOM network functions unusable

enesbcs, 2020-11-21:
I've done several tests on multiple ESP32 boards with LAN8720 (integrated and external also)- ESPIDF 3.3 and MicroPython 1.13:

On ESP32-WROVER with 4MB PSRAM:
-WiFi STA only: works OK
-LAN8720 only: works OK
-WiFi STA + LAN8720 at the same time works OK

On ESP32-WROOM with no PSRAM:
-WiFi STA only: works OK
-LAN8720 only: works OK
-WiFi STA + LAN8720 activated at the same time -> wired connection is available but webserver stops responding after a time anyway - [Errno 110] ETIMEDOUT
wifi sta connection get IP address but webserver is not available on that IP, the log contains some errors like:
 
I (151110) wifi:bcn_timout,ap_probe_send_start
I (153610) wifi:ap_probe_send over, resett wifi status to disassoc

Is it an ESPIDF limitation or may be patched in micropython somehow?

jonnor, 2024-10-01:
Does this problem still appear on recent MicroPython, version 1.23 or later?

jonnor, 2025-03-03:
Missing information to reproduce.

closed 2025-03-03 by jonnor