micropython/micropython #6648: BLE: Intermittent start up failure with new scheduled nimble stack

andrewleech, 2020-11-25:
Seg fault seen on unix port with debugger attached. Only happened once to me so far, re-running it operates as normal.

See image for more details, but basically it looks like some event is being handled before the conn struct has been fully initialised (it's still null pointer as seen in top left variables)
[image]

dpgeorge, 2020-11-25:
It looks like the code went through `ble_hs_conn_find_assert()` which would have asserted if BLE_HS_DBG_ASSERT was enabled.

andrewleech, 2020-11-25:
I'd say it's just a bit of a race condition between when the scheduled events are allowed to start processing (I think that's based on a a flag we set) and when the stack internals are properly ready. Hopefully not too hard to clean up, but I want to get the pairing squared away first.

jonnor, 2024-09-13:
Is this problem still relevant @andrewleech ?

andrewleech, 2024-09-13:
Hi @jonnor in suspect it still will be present however Bluetooth on the Unix port isn't regularly used so likely very rare. I myself haven't been using BLE for a couple of years at least.

The USB/serial Bluetooth dongle is serviced in a thread on Unix iirc so it's probably due to some parts of nimble being serviced there directly too early in this case.

Considering I haven't investigated this myself in 4 years I really shouldn't say I'll be looking at it again any time soon. From that angle it's reasonable to close this out. 
However this could also be considered valuable debug information on a hard to capture / intermittent segfault so I'm loath to lose the info.