micropython/micropython #6649: PULL Request with BASIC AUTH not working

FalconBFR, 2020-11-25:
Dear all,

I am trying to implement a simple pull request with basic auth on the esp32. As seen at https://github.com/micropython/micropython-lib/pull/311, it seems like it is implemented in urequests.pull and can be called by: `res = urequests.post(request_url, headers={'content-type': 'application/json'}, data=post_data, auth=('username','password')).json()
        print(res.text)`. 

However it seems like this doesn't work. Am I not calling this function right? There seems to be no documentation on requests in the official documentation.

Sincerely with Many Thanks,
Clement

SciWax, 2021-01-27:
Seems like another person has the same issue as you: https://github.com/micropython/micropython-lib/pull/310#issuecomment-653308187

Did you find a solution? I would be interested in one too.

SpotlightKid, 2021-04-22:
Actually, #311 is the PR (still open) implementing basic, which actually works.

See also: https://github.com/SpotlightKid/micropython-stm-lib/tree/master/mrequests

FalconBFR, 2021-04-24:
@SpotlightKid thanks for the reply and help. I will try it out but I won't be free to do so for at least until Mid-May due to other commitments. Hopefully it will work. Thanks!

Etruscian, 2021-08-19:
What is the status of this pull request? I managed to work around it by manually supplying the auth headers, but it would be nice to have it added to the module

hetvishah08, 2021-12-21:
Hi,

I also was facing the same issue for GET requests and I used this library, renamed it, and stored it on the board. This has Basic Authentication implemented and it worked fine for me.
[https://github.com/micropython/micropython-lib/blob/46e0fa18e76af9ce571d97590874282847d8e60f/urequests/urequests.py](url)

jonnor, 2024-09-13:
Basic auth has been implemented in https://github.com/micropython/micropython-lib/pull/311 for requests. And there are also other libraries, like the linked mrequests. So I believe we can close this.
For HTTP Digest, please see this thread: https://github.com/orgs/micropython/discussions/15138

closed 2024-09-13 by jonnor