micropython/micropython #6653: DEBUG=1 builds overflow the flash

iabdalkader, 2020-11-26:
```bash
make -j8 BOARD=NUCLEO_H743ZI DEBUG=1
```
```
CC build-NUCLEO_H743ZI/frozen_content.c
LINK build-NUCLEO_H743ZI/firmware.elf
arm-none-eabi-ld: build-NUCLEO_H743ZI/firmware.elf section `.text' will not fit in region `FLASH_TEXT'
arm-none-eabi-ld: region `FLASH_TEXT' overflowed by **3307612 bytes** <------- 
make: *** [Makefile:667: build-NUCLEO_H743ZI/firmware.elf] Error 1
```

dpgeorge, 2020-11-26:
> Seems the issue is with this `MICROPY_ROM_TEXT_COMPRESSION` if set to 0 the debug build generates reasonable binaries.

Yes, it uses a huge `?:` expression to do the compression.  Using `-O1` optimises this away and gets it to fit.  So does `MICROPY_ROM_TEXT_COMPRESSION=0`.  Probably the latter makes the most sense for debug builds.

jimmo, 2020-11-27:
I think disabling `MICROPY_ROM_TEXT_COMPRESSION` in debug builds makes sense, however I almost always use `-Og` with debug builds, which does handle this particular optimisation.

I would advocate for changing stm32/Makefile to make -Og the default for DEBUG instead of -O0.

jimmo, 2020-11-27:
FWIW, as well as the obvious size benefit, -Og is also a more "realistic" build. There's been too many times where -O0 makes the problem I'm investigating disappear.

iabdalkader, 2020-11-27:
@dpgeorge @jimmo I may switch to -Og as well eventually but I sent a PR to disable text compression when DEBUG=1 anyway. There's another important issue with debug builds on the stm32 ports and `PENDSV_DEBUG`, please see my detailed comment here on this old issue:
https://github.com/micropython/micropython/issues/315#issuecomment-577692954

Note I fixed this issue in our fork, but `PENDSV_DEBUG` is still used in the stm32 port.

dpgeorge, 2020-11-27:
> I would advocate for changing stm32/Makefile to make -Og the default for DEBUG instead of -O0.

That sounds good, except it actually leads to some warnings (`-Werror=maybe-uninitialized`).  So that would need a separate PR to get it all working correctly.

iabdalkader, 2020-11-29:
I have one more note about debugging, the following is needed to debug the H7 otherwise the DBGMCU gets disabled in sleep mode, ex. when calling WFI(), since you now use the template system.c I didn't know where to add this and send PR, it's also possible to keep DBGMCU running in stop, and standby modes.

```c
    #if defined(MCU_SERIES_H7) && (DEBUG==1)
    // Enable the Debug Module during Domain1/CDomain SLEEP mode
    DBGMCU->CR |= DBGMCU_CR_DBG_SLEEPD1;
    #endif
```

embeddedt, 2020-12-03:
FWIW I ran into a similar issue while working on `lv_micropython` STM32 support; is the plan still to make `-Og` the default eventually?

dpgeorge, 2020-12-04:
> I didn't know where to add this and send PR, it's also possible to keep DBGMCU running in stop, and standby modes.

It could probably go in system_stm32.c:SystemClock_Config().

>  is the plan still to make `-Og` the default eventually?

Yes, it needs a separate PR and some extra work to eliminate the warnings.

dpgeorge, 2020-12-07:
Overflow should be fixed by ce9197eb209bacd1453ae2ef83523aa0fa1f9b45

I'll leave this open until -Og is used in debug builds on stm32.

dpgeorge, 2020-12-07:
> I'll leave this open until -Og is used in debug builds on stm32.

This was easier than I thought, see #6682

dpgeorge, 2020-12-07:
> is the plan still to make `-Og` the default eventually?

Now done in cb1bb7592e18a50f33d22f84614c5ee9f45ce21e

closed 2020-12-07 by dpgeorge