micropython/micropython #6656: BLE disable / enable not working properly on unix port

andrewleech, 2020-11-27:
I tried to add a disable/re-enable after disconnect in an example test script 
```
    def _irq(self, event, data):
        ...
        elif event == _IRQ_CENTRAL_DISCONNECT:
            conn_handle, _, _ = data
            micropython.schedule(self.on_disconnect, None)

    def on_disconnect(self, *_):
        self._ble.active(False)
        time.sleep(1)
        self._ble.active(True)
        time.sleep(1)
        self._advertise()
```
This failed on the `self._ble.active(True)` line with a `OSError: [Errno 110] ETIMEDOUT`

I'll investigate this later once other BLE issues are out of the way

dpgeorge, 2020-11-27:
I think the issue is that you cannot run `ble.active()` in a schedule context, because that runs at a higher priority than NimBLE.

andrewleech, 2020-11-27:
Ah, ha yes of course, thanks for that. On my other platform this call is done from a usb command, so not from the ble callbacks / schedule.
Is there a way for code to detect its in schedule so that a clearer exception could be raised?

jonnor, 2024-09-13:
The main problem here seems to have been resolved. Perhaps we can close this issue?

closed 2024-09-14 by andrewleech