micropython/micropython #6660: ble_temperature_central.py  ENOMEMs after 1200 reads (MicroPython v1.13 on 2020-09-02; ESP32 module (spiram))

aguaviva, 2020-11-28:
My code has minor modifications to read a sensor from a different device.

Here is the error message:

> Traceback (most recent call last):
>   File "<stdin>", line 283, in <module>
>   File "<stdin>", line 271, in demo
>   File "<stdin>", line 223, in read
> OSError: [Errno 12] ENOMEM

this is the code

```
    # Issues an (asynchronous) read, will invoke callback with data.
    def read(self, callback):
        if not self.is_connected():
            return
        self._read_callback = callback
        self._ble.gattc_read(self._conn_handle, self._value_handle)  #<---------------------------here
```

aguaviva, 2020-11-28:
ok, using gc.collect() fixes the "mem leak"

aguaviva, 2020-11-28:
Reopening as I am still getting the ENOMEM after 1200 reads

jonnor, 2024-09-13:
Hi @aguaviva . Is this problem still present in recent MicroPython versions, such as 1.23 ? If so, please provide complete code to reproduce. Otherwise we cannot debug the problem.

aguaviva, 2024-09-14:
I moved to circuitpython, there things seem to work and issues get
addressed promptly