micropython/micropython #6663: RFC LCD160CR hardware/firmware

peterhinch, 2020-11-30:
This RFC is intended to create a record of issues I encountered interfacing an LCD160CR module with its official driver to ESP32. This was successful, enabling my [GUI](https://github.com/peterhinch/micropython-lcd160cr-gui) to run.

These notes might be of interest if a future hardware update occurs. Item 4 is a docs issue.
 1. I experienced sporadic initial I2C connection problems until I implemented code to assert `PWR` followed by a delay before instantiating the class. This implies that [this delay](https://github.com/micropython/micropython/blob/f7225d1c9527c9839d80b8f73641ddb4ebbe6105/drivers/display/lcd160cr.py#L56) should be increased. I can't claim this as proven, but I have had no issues since implementing this workround.
2. The SPI interface appears to have no support for `SS/`. This precludes SPI bus sharing.
3. The black connector `P2` at the end of the PCB claims to support SPI, however the V1.0 schematic shows it being connected to different pins on the chip compared with the other SPI connections (`SPI1_CLK` vs `SPI0_CLK`, similar for `MOSI`). I don't know the reasoning here, but the SPI interface on the `P2` connector doesn't seem to work.
4. The [docs](http://docs.micropython.org/en/latest/library/lcd160cr.html) provide links to a datasheet and schematic which are out of date. The current issue of both should be V1.1.

jonnor, 2024-08-18:
Are these issue still relevant, or can we close this issue?

peterhinch, 2024-08-19:
Closing. LCD160CR is long gone, sadly :(

closed 2024-08-19 by peterhinch