micropython/micropython #6670: JavaScript port utime is incomplete

amirgon, 2020-12-02:
[utime implementation](https://github.com/micropython/micropython/blob/c8b055717805500494a14870a4200bbc933fe337/ports/javascript/modutime.c#L38-L49) on the JS port is missing functions related to real time clock, such as:  
`clock`, `time`, `time_ns`, `gmtime`, `localtime`, `mktime`

amirgon, 2021-01-09:
Missing modules in the JS port: `time`, `select`.
These are apparently needed for uasyncio in the JS port.

Related: https://github.com/lvgl/lvgl/issues/1320#issuecomment-757361304

jonnor, 2024-08-18:
Is this still the case in the latest MicroPython (1.23)? There has been considerable work on the JS port (now called `webassembly`) lately.

dpgeorge, 2024-08-18:
The webassembly port now has `time.gmtime`, `time.mktime`, `time.localtime`, `time.time` and `time.time_ns`.  It also has a custom `asyncio` implementation that allows top-level await.

closed 2024-08-18 by dpgeorge