micropython/micropython #6672: stm32: USB CDC driver does not detect abrupt USB disconnection

dpgeorge, 2020-12-03:
If a USB cable is unplugged while a serial terminal program is still active (on the host side) then the USB CDC driver on the stm32 device still thinks that it is connected.  This means that any data written to the USB CDC will block for a period of time, because it's waiting for the host to receive that data.

In practice this mean that a device can become very slow (waiting for serial to drain) if a USB REPL connection is pulled out without closing the serial terminal program.

To reproduce on a NUCLEO_WB55, connect to the ST-link USB port (via USB-UART) and run the following:
```python
import pyb, time
while 1:
    pyb.LED(2).toggle()
    pyb.LED(3).intensity(pyb.USB_VCP(0).isconnected())
    print('test' * 50)
    time.sleep(0.25)
```
The green LED will blink.  Then connect a serial program to the other USB port and the blue LED should turn on.  Then pull out the second USB port and the blue LED will stay on and eventually the green LED will stop blinking.

hoihu, 2020-12-09:
Is this something that has been introduced recently?

dpgeorge, 2020-12-10:
No, I think it's been there all along.  But it doesn't seem to affect MCUs like F4 and F7.

dpgeorge, 2021-02-02:
The reason this bug affects WB MCU and not F4/F7 is because WB does not implement the `HAL_PCD_DisconnectCallback` event.

dpgeorge, 2021-02-02:
Fixed by 7be1f7790278d2c40e7a828e340c43f257c7d748

closed 2021-02-02 by dpgeorge