micropython/micropython #6675: boot loop after fresh firmware install on ESP8266MOD 12-F and ESP-12E

nithir, 2020-12-03:
I buy some nodeMCU on ebay, 
i load the last micropython (1.12)... at boot i saw REPL and after 8s the device boot.
I find some info in the forum ([here](https://forum.micropython.org/viewtopic.php?f=16&t=6092&start=10))... I have to run this : 

`import network; network.WLAN(network.AP_IF).active(False)`

and it works very well. All my board could actually work with micropython if i disable wifi AP mode at start.

i try STA mod and it works well but if i try AP mod, the boot loop start again...

So i tried to flash with arduino an AP mode test program and ... It works !

Obviously there is somethings wrong with this board when AP mode start with micropython.
Some people in the [forum](https://forum.micropython.org/viewtopic.php?f=16&t=7725) says that it could be a lack of power and to solve that, put a 100µF cap between NE and ground... i don't try this because it works with arduino without it... That's why i think that it's not an hardware problem.

is someone interested by my problem ?

bg1000, 2021-02-19:
This issue is still present in 1.14. @nithir I seem to have only a fraction of a second to respond.  How were you able to run ```import network; network.WLAN(network.AP_IF).active(False) ```?

nithir, 2021-02-19:
This command stop the boot loop.
It's a first step. I could use it in STA mode but in AP mode it start to boot loop again.
Thank your for your help.

bg1000, 2021-02-19:
@nithir - My question is what method did you use to get the ESP to accept this command. Mine reboots almost immediately and I can not get it to accept any commands via terminal.

nithir, 2021-02-19:
excuse me, english is not my first language...
To enter the command, I paste in a text editor and I add a new line like that :
```
 import network; network.WLAN(network.AP_IF).active(False)

```
and I copy that in the clipboard with the new line character at the end in order to paste it in a terminal with the new line character.

I connect serial tool like picocom (linux) in order to watch micropython output.
when I see REPL prompt I paste my clipboard. You can paste it in loop. 
The command execute and you get a new prompt but this time no loop...

bg1000, 2021-02-19:
@nithir - I tried something similar but I don't think I was inserting the newline character properly. Thanks for replying. I'll go back and try again.

bg1000, 2021-02-20:
I wanted to provide an update for anyone who might be having this issue and finds this.  Unlike @nithir I was unable to get the copy/paste method to work (maybe I didn't get the timing exactly right).  Here is an alternate method that worked for me:

1. Save ```import network; network.WLAN(network.AP_IF).active(False)``` in a file called boot.py
2. Install ampy with ```pip3 install aradfruit-ampy```
3. Place the following on the terminal prompt without pressing the enter key ``` ampy --port /dev/ttyUSB0 put boot.py```
4. Press and hold down the reset button on the node MCU.
5. Release the reset button and immediately press the enter key on the keyboard.

Timing is critical as it seems that you have only a fraction of a second after the reset to initiate file transfer. In order to get it to work I had to use one hand for the reset button and the other hand for the enter key.

Rick7W, 2021-05-30:
My background is design of switching power supplies, analog  signal processing & DSP, automotive embedded controllers, automation/motion control, etc.  I spent several hours testing to figure out what the core issue is and then found it in the AI-Thinker ESP-12F datasheet.  There is supposed to be a 10 microfarad (10 mfd or 10 uF) capacitor from 3.3 volts terminal to ground in addition to a 100 nanofarad (100 nF or 0.1 microfarad often marked "104" for 100,000 picofarad) capacitor for energy storage high frequency bypass respectively.  Apparently, the quick pulses of current that the ESP8266 draws in certain modes temporarily overload the ability for the 3.3 volt regulator to supply enough current...which is probably aggravated by the leakage inductance in the USB cable.  I tried adding a 100 microfarad capacitor from 5 volt to ground with 100 nanofarad caps on both the reset and enable terminals to ground and they helped but did not fix it 100%.  Only after I added the 10 microfarad (tantalum) from 3.3v to ground was the problem completely eradicated and I was then able to remove the two 100 nanofarad (mono ceramic) caps and the 100 microfarad from reset, enable and 5 volt respectively.  Which is exactly what the AI-Thinker datasheet shows.  Why it was left off some of the cheaper D1 mini knock-off modules I can only assume because of cost and size issues.  To keep the ESR & physical size down I used a tantalum capacitor.  Triple check to make sure you get the polarity correct and use at least 10 volt rated or greater.  BTW, I tried it with a 4.7 microfarad tantalum and that worked fine too...but the manufacturer spec'd 10 microfarad.  I would suggest avoiding general purpose electrolytic caps because the ESR & ESL (Equivalent Series Resistance, Equivalent Series Inductance) is pretty high for the little ones especially at high frequencies.  Low ESR electrolytic caps may work fine but may be more difficult to obtain than the tantalum.  I have LCR analyzers and checked the ESR before installing each cap.  IMO, it would be good practice to also add a 22-100 microfarad tantalum or low ESR electrolytic from 5 volts to ground which would help compensate for lesser quality USB cables.  If used in a noisy EMI/RFI environment (such as around a switching power supply/regulator IC) then adding the two 100 nanofarad capacitors from Reset and Enable to ground...may help prevent sporadic issues.  Hope this helps.

Hevienz, 2021-09-23:
Still occurs on esp8266-20210902-v1.17.bin.

nithir, 2021-09-23:
If its a power problem caused by the design... I don't understand why AP mode works well with Arduino firmware.. .

tyfncn, 2023-08-11:
This seems an old issue but I've also experienced exatly same as @nithir mentioned on MicroPython v1.20.0 three different esp boards; 
-  esp01 1M module 8 pins + ch340 debugger with big cap
- new NodeMCU v3 labelled board
- esp12e + ch340 debugger chip

Since ch340 debugger had a cap already, I've tested nodemcu board with 4 tantalum caps as @Rick7W  described but no success (reset time increased from 8 to 30 seconds at one test).

I've found a workaround by looking at used channels with `ap.scan()` or `Wifi Analyzer` on android; it turns out default config enables ap on channel 1 and it's pretty noisy on around me. Thus configuring ap with another channel as `ap.config(essid="pyboard", password="aabbccdd11", security=network.AUTH_WPA_WPA2_PSK, channel=11)` fixes the situation.

With arduino ide `ESP8266WiFi/WifiAccessPoint.ino` has no problem and Micropython serial has `wdt reset`  line  hints that a task takes long and esp watchdog timer resets the chip. But I couldn't debug further into internals to be sure.

jonnor, 2024-09-13:
Thanks a lot of reporting workarounds here, that is very helpful to other users.