micropython/micropython #6691: How to properly lock / make set of operations atomic? Mutexes created via `_thread.allocate_lock()` create deadlocks...

mirko, 2020-12-11:
I'm using the micropython-lib/mqtt library on an ESP32 and am experiencing race conditions.
I narrowed them down to an interference between an ongoing MqttClient.publish() and an MqttClient.publish() triggered by a timer.
Makes sense, an ISR is interrupting an ongoign read/write from/to the socket - so let's make things atomic.
I tried so by guarding the code with mutexes, in this case locks created by `_thread.allocate_lock()`.
However here I manage to get myself into a deadlock.
Here's a minimal code snippet causing a deadlock which I wouldn't have expected:
```

from micropython import schedule
from machine import Timer
import _thread
import utime

lock = _thread.allocate_lock()

def do_stuff():
    print("==", utime.time(), "==", "do_stuff() called")
    with lock:
        print("do_stuff(): lock acquired")
        utime.sleep(1)
    print("==", utime.time(), "==", "do_stuff(): lock released")

def do_other_stuff(_=None):
    print("==", utime.time(), "==", "do_other_stuff() called")
    with lock:
        print("do_other_stuff(): lock acquired")
        utime.sleep(1)
    print("==", utime.time(), "==", "do_other_stuff(): lock released")

tmr = Timer(0)
tmr.init(mode=Timer.ONE_SHOT, period=500, callback=lambda t: schedule(do_other_stuff, None))
do_stuff()
```

The `schedule()` within the callback is not really necessary, as on ESP32 it's already a soft IRQ anyway (l138: `     mp_sched_schedule(self->callback, self);` in `ports/esp32/machine_timer.c`.

But it's not only the timer, lot's of interrupts trigger mqtt messages being published - handlers attached to GPIOs (also soft IRQs only) and so on.

Question 1: Why is this causing a deadlock? The code trying to acquire the lock is scheduled for being executed outside the ISR.
Question 2: How to ensure execution of code is atomic if not via mutexes created via the `_thread`-module?

dlech, 2020-12-11:
Both functions are probably being run on the same thread. `utime.sleep(1)` will call `MICROPY_EVENT_POLL_HOOK` which calls the scheduled function while the lock is still held, resulting in the deadlock.

So instead of using a `Timer()` to defer `do_other_stuff()`, you could add `utime.sleep(0.5)` to the start of `do_other_stuff()` and then run it a different thread.

mirko, 2020-12-11:
> So instead of using a Timer() to defer do_other_stuff(), you could add utime.sleep(0.5) to the start of do_other_stuff() and then run it a different thread.

This is only a minimal example. In reality the timer is configured as periodic and firing (and publishing on the MQTT bus) every `n` seconds. So, in case I don't misunderstand, I don't think this would work as a workaround.

dlech, 2020-12-11:
So in that case, maybe something like this...

```python
def background_thread():
    while True:
        utime.sleep(n)
        with lock:
            fire()
```

But the main point is that regardless of the actual algorithm, thread locks are only going to work if they are used on different threads.

For locking with respect to interrupts, interrupts should be disabled in the critical sections of code that access the same data as the interrupt handler.

mirko, 2020-12-11:
> But the main point is that regardless of the actual algorithm, thread locks are only going to work if they are used on different threads.
> For locking with respect to interrupts, interrupts should be disabled in the critical sections of code that access the same data as the interrupt handler.

How would I do that inside micropython, preferably platform/port independant?

dlech, 2020-12-11:
`machine.irq_disable()` ?

https://docs.micropython.org/en/latest/library/machine.html#machine.disable_irq

mirko, 2020-12-11:
> `machine.irq_disable()` ?
> 
> https://docs.micropython.org/en/latest/library/machine.html#machine.disable_irq

Oh, ok, thanks! I'll definitely try this.
Another advantage I was hoping to get for free when using `with lock:` was, that operations which can't acquire the lock immediately are going to be queued and executed later, once the lock was free'd. When disabling interrupts everything happening in this time is being discarded as far as I understand, right?

dlech, 2020-12-11:
The interrupt handlers will just not run until irqs are enabled again. But the interrupt will be raised as soon as interrupts are enabled again - it won't be lost. There is a potential for bad things to happen,l e.g. buffer overruns in UARTs, if interrupts are disabled for too long. So generally they are only disabled for the absolute critical section where you read or write the one variable share with the interrupt handler.

mirko, 2020-12-11:
> The interrupt handlers will just not run until irqs are enabled again. But the interrupt will be raised as soon as interrupts are enabled again - it won't be lost. 

Good to know, thanks!

> There is a potential for bad things to happen, e.g. buffer overruns in UARTs, if interrupts are disabled for too long. So generally they are only disabled for the absolute critical section where you read or write the one variable shared with the interrupt handler.

And apparently writing onto a TCP socket (my usecase: making MQTT publish() calls not interfering with each other) is way too much / too unpredictable. Watchdog triggers reset before interrupts are being re-enabled (`Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)`). I can now obviously also disable the watchdog for that section, but I'm getting more and more the feeling this road is a dead end (or one without any end..)(?)