micropython/micropython #6692: rtc.c:153:17: error: implicit declaration of function 'LL_RTC_GetSynchPrescaler'

neurotoxin78, 2020-12-11:
I`m try to build for stm32f411CE and get an error
rtc.c: In function 'rtc_init_start':
rtc.c:153:17: error: implicit declaration of function 'LL_RTC_GetSynchPrescaler' [-Werror=implicit-function-declaration]
  153 |             if (LL_RTC_GetSynchPrescaler(RTC) != RTC_SYNCH_PREDIV
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~
rtc.c:154:20: error: implicit declaration of function 'LL_RTC_GetAsynchPrescaler' [-Werror=implicit-function-declaration]
  154 |                 || LL_RTC_GetAsynchPrescaler(RTC) != RTC_ASYNCH_PREDIV) {
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~
rtc.c:156:17: error: implicit declaration of function 'LL_RTC_DisableWriteProtection' [-Werror=implicit-function-declaration]
  156 |                 LL_RTC_DisableWriteProtection(RTC);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
rtc.c:157:17: error: implicit declaration of function 'LL_RTC_EnableInitMode'; did you mean 'RTC_EnterInitMode'? [-Werror=implicit-function-declaration]
  157 |                 LL_RTC_EnableInitMode(RTC);
      |                 ^~~~~~~~~~~~~~~~~~~~~
      |                 RTC_EnterInitMode
rtc.c:160:25: error: implicit declaration of function 'LL_RTC_IsActiveFlag_INIT' [-Werror=implicit-function-declaration]
  160 |                     if (LL_RTC_IsActiveFlag_INIT(RTC)) {
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~
rtc.c:162:25: error: implicit declaration of function 'LL_RTC_SetSynchPrescaler' [-Werror=implicit-function-declaration]
  162 |                         LL_RTC_SetSynchPrescaler(RTC, RTC_SYNCH_PREDIV);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~
rtc.c:163:25: error: implicit declaration of function 'LL_RTC_SetAsynchPrescaler' [-Werror=implicit-function-declaration]
  163 |                         LL_RTC_SetAsynchPrescaler(RTC, RTC_ASYNCH_PREDIV);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~
rtc.c:164:25: error: implicit declaration of function 'LL_RTC_DisableInitMode' [-Werror=implicit-function-declaration]
  164 |                         LL_RTC_DisableInitMode(RTC);
      |                         ^~~~~~~~~~~~~~~~~~~~~~
rtc.c:168:17: error: implicit declaration of function 'LL_RTC_EnableWriteProtection' [-Werror=implicit-function-declaration]
  168 |                 LL_RTC_EnableWriteProtection(RTC);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:77: build-WeAct_F411CE/rtc.o] Error 1

What to do?

ru-sh, 2020-12-17:
Switching to the tag `v1.12` resolved the issue.

dpgeorge, 2020-12-18:
It probably needed a `git submodule update`.

closed 2020-12-18 by dpgeorge

neurotoxin78, 2020-12-18:
> 
> 
> Switching to the tag `v1.12` resolved the issue.

Yes thank you.