micropython/micropython #6698: [STM32WB] machine.freq set not implemented

dmazzella, 2020-12-14:
Hi, 
is there a plan to support the frequency change for the stm32wb?

Thanks and regards, 
D.

dmazzella, 2021-04-21:
update on this?

dpgeorge, 2021-07-26:
See #7588 for this feature.

dmazzella, 2021-07-31:
I'm on vacation, as soon as I get back I take the tests and update you.

dmazzella, 2021-08-04:
Hi @dpgeorge,

I started testing, the frequency seems to change correctly with all supported values.

the first test was a simple loop with a led toggle every second and this worked with all frequencies.

the second test is with aioble the basic test [temp_sensor.py](https://github.com/micropython/micropython-lib/blob/master/micropython/bluetooth/aioble/examples/temp_sensor.py) with the only addition of the frequency change before `asyncio.run(main())`

```python
machine.freq(32_000_000)
asyncio.run(main())
```

 and for all frequencies below 64_000_000 the peripheral is visible but does not allow me to connect

```python
MicroPython v1.16-160-ga3675294a-dirty on 2021-08-04; LEMMEEN-WB55 with STM32WB55CGU6
Type "help()" for more information.
>>> import stm
>>> stm.rfcore_fw_version(0)
(1, 1, 0, 0, 0)
>>> stm.rfcore_fw_version(1)
(1, 11, 0, 0, 1)
>>> 
```

dmazzella, 2021-08-04:
updated rf firmware to latest version work at 32_000_000 but not at lowest frequencies

```python
>>> stm.rfcore_fw_version(0)
(1, 2, 0, 0, 0)
>>> stm.rfcore_fw_version(1)
(1, 12, 0, 0, 7)
```

dpgeorge, 2021-08-05:
> updated rf firmware to latest version work at 32_000_000 but not at lowest frequencies

Thanks for testing.  I don't know how the RF core works when the frequency is not 64MHz and did not test this.  I was mainly interested in supporting the low-power-run mode to start with.  Additional work will need to be done to support BLE with a lower CPU frequency (if it's possible at all).

dmazzella, 2021-08-05:
ok, low-power-run mode work correctly with all frequencies supported.

closed 2021-08-07 by dmazzella