micropython/micropython #6700: Impossible to display degree symbol

bill-orange, 2020-12-16:
There are several ways in regular Python to display the degree symbol.  You can simply insert  u"\N{DEGREE SIGN}" .  In micropython this appear to be unsupported.  Sometimes you can cut and paste in °.  Very simple, but in micropython this results in a funny A  followed by °.  Several other techniques have the same effect, funny A with degree symbol.

Is adding the degree symbol support a planned addition to micropython and for now is there a work-around?

dpgeorge, 2020-12-17:
>  You can simply insert u"\N{DEGREE SIGN}"

Correct, unicode names are not supported (it would take many megabytes to store the mapping from name to unicode point/number).

> Is adding the degree symbol support a planned addition to micropython and for now is there a work-around?

I'm pretty sure it's supported, it is in the end just a unicode character.  Eg:
```python
degree = str(b'\xc2\xb0', 'utf8') # create from bytes
degree = chr(176) # create from known unicode point
sys.stdout.write(degree)
```
That will print the degree symbol.

bill-orange, 2020-12-17:
@dpgeorge thanks for the response.  I tested your solutions with this result. 

`u"\N{DEGREE SIGN}"`  results in the error NotImplementedError: unicode name escapes

degree = chr(176) results in `Â°` when output to a web page through html.  It works on the command line.

Both approaches work fine in regular Python.

dpgeorge, 2020-12-18:
> `u"\N{DEGREE SIGN}"` results in the error NotImplementedError: unicode name escapes

Yes, what I meant above is that this form is not supported in MicroPython.  So you cannot do it this way.

> degree = chr(176) results in `Â°` when output to a web page through html.

How exactly is it being output to a web page, what is the code that reproduces the problem?

bill-orange, 2020-12-18:
I moved a copy of my files to GitHub.  Take a look at BME280.py line 273 .  To run this you would need to put your username and password in boot.py.

https://github.com/bill-orange/MicroPython/tree/main/BME280%20test

dpgeorge, 2020-12-18:
Thanks for posting the code.

I can't seem to reproduce any issue.  On the unix port of MicroPython I can do:
```
$ micropython
MicroPython v1.13-186-g5a7027915-dirty on 2020-11-27; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> s="{}.{:.01d}{}C".format(1, 2, chr(176))
>>> s
'1.2\xb0C'
>>> print(s)
1.2°C
>>> 
```
Turning that into a HTML, you may need to use `&deg;`?

bill-orange, 2020-12-18:
Yes, in the command line it works for me also.  I will try your suggestion tomorrow.

From: Damien George 
Sent: Thursday, December 17, 2020 6:25 PM
To: micropython/micropython 
Cc: bill-orange ; Author 
Subject: Re: [micropython/micropython] Impossible to display degree symbol (#6700)

Thanks for posting the code.

I can't seem to reproduce any issue. On the unix port of MicroPython I can do:

$ micropython

MicroPython v1.13-186-g5a7027915-dirty on 2020-11-27; linux version

Use Ctrl-D to exit, Ctrl-E for paste mode

>>> s="{}.{:.01d}{}C".format(1, 2, chr(176))

>>> s

'1.2\xb0C'

>>> print(s)

1.2°C

Turning that into a HTML, you may need to use &deg;?

bill-orange, 2020-12-18:
That worked. Thanks.  

In main.py

`<tr><td>Temp. Celsius</td><td><span class="sensor">""" + str(bme.temperature) + """&degC</span></td></tr>`

closed 2020-12-18 by bill-orange

DrKRR, 2023-05-23:
The following MicroPython Program displays temperature in degree Celsius (symbol) on OLED
import ssd1306
from machine import Pin, SoftI2C
import time
import framebuf

i2c = SoftI2C(scl=Pin(22), sda=Pin(21))
oled = ssd1306.SSD1306_I2C(128, 64, i2c)

# Custom character bitmap
custom_bitmap = bytearray([0x00, 0x0e, 0x11, 0x11, 0x0e, 0x00, 0x00, 0x00])

# Create a frame buffer from the custom bitmap
fb = framebuf.FrameBuffer(custom_bitmap, 8, 8, framebuf.MONO_HLSB)
t= b'45.2'
t1= t.decode()
# Display custom bitmap on OLED
oled.fill(0)  # Clear the display

# Blit the frame buffer to the OLED display at position (0, 0)
oled.blit(fb, 95, 0)
oled.text("Temper= " + t1 + " " + "C", 0, 5)
# Update the display
oled.show()

bruno1950, 2024-04-20:
Hi, I'm an old (74) newbie of micropython and ESP32 and I'm trying to use an OLED SH1106 to display data from a BME280 sensor connected to an ESP32 working with uPY. While on the serial monitor the ° degree symbol is correctly printed I was not able to get it displayed on the SH1106. I tried both display.text("{:0.0f}°C".format(temperature), 0, 30) and display.text("{:0.0f}{}".format(temperature, chr(176)), 0, 30) but I keep on getting ValueError: unknown format code 'f' for object of type 'str'. Could you hel pme to fix this issue ? Thanks

peterhinch, 2024-04-21:
The solution depends on which display driver you're using. [This one](https://github.com/peterhinch/micropython-nano-gui/blob/master/drivers/sh1106/sh1106.py) it is subclassed from [FrameBuffer](https://docs.micropython.org/en/latest/library/framebuf.html): this enables you to use fonts created with [font_to_py](https://github.com/peterhinch/micropython-font-to-py). Such fonts can contain any glyph you like.

bruno1950, 2024-04-21:
Hi,

thanks for your help that I really appreciate since I'm an old (74) newbie struggling with micropython and I've found your name associated with an amazing mass of python codes and libraries. I substituted the sh1106.py with the one you suggested but I get the error message File "sh1106.py", line 78, in <module>
ImportError: no module named 'drivers'. When I checked if there was a drivers.py module available (I'm using Thonny, since I am a beginner) I just found this link for download https://pypi.org/project/drivers/ nut I do not know if it's the right reference

peterhinch, 2024-04-21:
The driver is designed to work as a Python package. It needs to be installed with `mpremote` as described [here](https://github.com/peterhinch/micropython-nano-gui/blob/master/DRIVERS.md#12-installation). That will create the right directory structure and install `boolpalette.py`.

bruno1950, 2024-04-21:
Thanks for your patient support..

I think that I have to read to read carefully the documentation you suggest before asking questions. The problem is that it's just few months that I shifted from arduino IDE to micropython and even if I tried to study Python the only way to learn is to read codes, try to understand them, make errors, ask for help and repeat all these steps. At the present the step where I'm more successful is to make errors.... Have a nice evening

peterhinch, 2024-04-21:
You won't regret the move. I ditched Arduino for MicroPython ten years ago and have never looked back.