micropython/micropython #6708: File modification dates are wrong on device

PEConn, 2020-12-19:
Hello,

I'm using `rshell` to interact with my device, but I *think* this seems like a core micropython problem (I could be wrong). Here's what's happening:

```shell
# Check the date of our file
> ls -l src/sayhello.py
410 Dec 19 08:52 src/sayhello.py

# Connect to device
> rshell --port /dev/ttyUSB0

# Copy the file over
> cp src/sayhello.py /pyboard

# Check the file modification date
> ls -l /pyboard/sayhello.py
410 Jan 26 2127  /pyboard/sayhello.py
```

My `rshell` version is `0.0.28` and running `import os;os.uname()` gives me:

```python
(
  sysname='esp8266',
  nodename='esp8266',
  release='2.2.0-dev(9422289)',
  version='v1.13 on 2020-09-11',
  machine='ESP module with ESP8266'
)
```

Maybe I've set something up wrong, any help would be appreciated!

peterhinch, 2020-12-19:
I use rshell V0.0.28 daily on Pyboards, ESP32 and ESP8266 and have never encountered this. I suggest you check the RTC time on the ESP8266:
```python
>>> from machine import RTC
>>> rtc=RTC()
>>> rtc.datetime()
(2020, 12, 19, 5, 11, 16, 52, 392834)
>>> 
```

PEConn, 2020-12-19:
Thanks for the suggestion, unfortunately it doesn't look like that's the problem - my RTC looks sane:

```python
>>> from machine import RTC
>>> rtc = RTC()
>>> rtc.datetime()
(2020, 12, 19, 5, 13, 22, 59, 373)
```

(Also, what is the `5` in both of our outputs? I [couldn't find it in the docs](https://docs.micropython.org/en/latest/library/machine.RTC.html).)

peterhinch, 2020-12-20:
> it doesn't look like that's the problem

Indeed. I'm foxed. The only thing I can think of (grasping at straws) is to erase flash and reinstall, but maybe you've already done that.
> Also, what is the 5 in both of our outputs?

Day of the week. Today you'll see 6. The `machine.RTC` documentation is poor.

PEConn, 2020-12-20:
I hadn't but I did try just now - right after flashing `ls -al /pyboard` gives me:

```
230 Jan 27 2127  boot.py
```

I wonder if it's something with the device.

jonnor, 2024-08-18:
Is this issue still present on latest MicroPython? @PEConn