micropython/micropython #6709: Building of example cpp usermodule fail for ESP32 port.

wgaylord, 2020-12-19:
I am working on making my own cpp module but I found that I can not even build the eample module without error.

The following is the error that I get when I attempt to build it.

```
chibill@jwg1:/data_pool/micropython/project/micropython/ports/esp32$ make clean
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Building with ESP IDF v4
rm -rf build-GENERIC
chibill@jwg1:/data_pool/micropython/project/micropython/ports/esp32$ make USER_C_MODULES=../../examples/usercmodule/ CFLAGS_EXTRA=-DMODULE_CPPEXAMPLE_ENABLED=1
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from ../../examples/usercmodule//cexample
Including User C Module from ../../examples/usercmodule//cppexample
Building with ESP IDF v4
GEN build-GENERIC/sdkconfig.h
mkdir -p build-GENERIC/bootloader//data_pool/micropython/esp/esp-idf//components/bootloader/subproject/main/
mkdir -p build-GENERIC/bootloader//data_pool/micropython/esp/esp-idf//components/bootloader_support/src/
mkdir -p build-GENERIC/bootloader//data_pool/micropython/esp/esp-idf//components/bootloader_support/src/esp32/
mkdir -p build-GENERIC/bootloader//data_pool/micropython/esp/esp-idf//components/log/
mkdir -p build-GENERIC/bootloader//data_pool/micropython/esp/esp-idf//components/soc/esp32/
mkdir -p build-GENERIC/bootloader//data_pool/micropython/esp/esp-idf//components/soc/src/
mkdir -p build-GENERIC/bootloader//data_pool/micropython/esp/esp-idf//components/spi_flash/
CC /data_pool/micropython/esp/esp-idf//components/bootloader_support/src/bootloader_clock.c
CC /data_pool/micropython/esp/esp-idf//components/bootloader_support/src/bootloader_common.c
CC /data_pool/micropython/esp/esp-idf//components/bootloader_support/src/bootloader_flash.c
CC /data_pool/micropython/esp/esp-idf//components/bootloader_support/src/bootloader_flash_config.c
CC /data_pool/micropython/esp/esp-idf//components/bootloader_support/src/bootloader_init.c
CC /data_pool/micropython/esp/esp-idf//components/bootloader_support/src/bootloader_random.c
CC /data_pool/micropython/esp/esp-idf//components/bootloader_support/src/bootloader_utility.c
CC /data_pool/micropython/esp/esp-idf//components/bootloader_support/src/flash_qio_mode.c
CC /data_pool/micropython/esp/esp-idf//components/bootloader_support/src/esp_image_format.c
CC /data_pool/micropython/esp/esp-idf//components/bootloader_support/src/flash_encrypt.c
CC /data_pool/micropython/esp/esp-idf//components/bootloader_support/src/flash_partitions.c
CC /data_pool/micropython/esp/esp-idf//components/bootloader_support/src/esp32/bootloader_sha.c
CC /data_pool/micropython/esp/esp-idf//components/bootloader_support/src/esp32/secure_boot.c
AR build-GENERIC/bootloader/libbootloader_support.a
CC /data_pool/micropython/esp/esp-idf//components/log/log.c
AR build-GENERIC/bootloader/liblog.a
CC /data_pool/micropython/esp/esp-idf//components/spi_flash/spi_flash_rom_patch.c
AR build-GENERIC/bootloader/libspi_flash.a
CC /data_pool/micropython/esp/esp-idf//components/soc/esp32/cpu_util.c
CC /data_pool/micropython/esp/esp-idf//components/soc/esp32/gpio_periph.c
CC /data_pool/micropython/esp/esp-idf//components/soc/esp32/rtc_clk.c
CC /data_pool/micropython/esp/esp-idf//components/soc/esp32/rtc_clk_init.c
CC /data_pool/micropython/esp/esp-idf//components/soc/esp32/rtc_init.c
CC /data_pool/micropython/esp/esp-idf//components/soc/esp32/rtc_periph.c
CC /data_pool/micropython/esp/esp-idf//components/soc/esp32/rtc_pm.c
CC /data_pool/micropython/esp/esp-idf//components/soc/esp32/rtc_sleep.c
CC /data_pool/micropython/esp/esp-idf//components/soc/esp32/rtc_time.c
CC /data_pool/micropython/esp/esp-idf//components/soc/esp32/rtc_wdt.c
CC /data_pool/micropython/esp/esp-idf//components/soc/esp32/sdio_slave_periph.c
CC /data_pool/micropython/esp/esp-idf//components/soc/esp32/sdmmc_periph.c
CC /data_pool/micropython/esp/esp-idf//components/soc/esp32/soc_memory_layout.c
CC /data_pool/micropython/esp/esp-idf//components/soc/esp32/spi_periph.c
CC /data_pool/micropython/esp/esp-idf//components/soc/src/memory_layout_utils.c
AR build-GENERIC/bootloader/libsoc.a
CC /data_pool/micropython/esp/esp-idf//components/bootloader/subproject/main/bootloader_start.c
AR build-GENERIC/bootloader/libmain.a
LINK build-GENERIC/bootloader.elf
Create build-GENERIC/bootloader.bin
esptool.py v2.8
Create build-GENERIC/partitions.bin
mkdir -p build-GENERIC/genhdr
GEN build-GENERIC/genhdr/mpversion.h
GEN build-GENERIC/genhdr/moduledefs.h
GEN build-GENERIC/genhdr/qstr.i.last
In file included from ../../py/mpconfig.h:62,
                 from ../../py/mpstate.h:31,
                 from ../../py/runtime.h:29,
                 from ../../examples/usercmodule//cppexample/examplemodule.h:2,
                 from ../../examples/usercmodule/cppexample/example.cpp:2:
./mpconfigport.h:5:10: fatal error: mpconfigboard.h: No such file or directory
 #include "mpconfigboard.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
Traceback (most recent call last):
  File "../../py/makeqstrdefs.py", line 188, in <module>
    preprocess()
  File "../../py/makeqstrdefs.py", line 62, in preprocess
    for output in p.imap(pp(flags), chunks):
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 868, in next
    raise value
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "../../py/makeqstrdefs.py", line 46, in run
    return subprocess.check_output(args.pp + flags + files)
  File "/usr/lib/python3.8/subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['xtensa-esp32-elf-gcc', '-E', '-std=gnu++11', '-Os', '-ffunction-sections', '-fdata-sections', '-fstrict-volatile-bitfields', '-mlongcalls', '-nostdlib', '-Wall', '-Werror', '-Wno-error=unused-function', '-Wno-error=unused-but-set-variable', '-Wno-error=unused-variable', '-Wno-error=deprecated-declarations', '-DESP_PLATFORM', '-I../../lib/berkeley-db-1.xx/PORT/include', '-I.', '-I../..', '-Ibuild-GENERIC', '-I/data_pool/micropython/esp/esp-idf//components/bootloader_support/include', '-I/data_pool/micropython/esp/esp-idf//components/bootloader_support/include_bootloader', '-I/data_pool/micropython/esp/esp-idf//components/console', '-I/data_pool/micropython/esp/esp-idf//components/driver/include', '-I/data_pool/micropython/esp/esp-idf//components/driver/include/driver', '-I/data_pool/micropython/esp/esp-idf//components/efuse/include', '-I/data_pool/micropython/esp/esp-idf//components/efuse/esp32/include', '-I/data_pool/micropython/esp/esp-idf//components/esp32/include', '-I/data_pool/micropython/esp/esp-idf//components/espcoredump/include', '-I/data_pool/micropython/esp/esp-idf//components/soc/include', '-I/data_pool/micropython/esp/esp-idf//components/soc/esp32/include', '-I/data_pool/micropython/esp/esp-idf//components/heap/include', '-I/data_pool/micropython/esp/esp-idf//components/log/include', '-I/data_pool/micropython/esp/esp-idf//components/nvs_flash/include', '-I/data_pool/micropython/esp/esp-idf//components/freertos/include', '-I/data_pool/micropython/esp/esp-idf//components/esp_ringbuf/include', '-I/data_pool/micropython/esp/esp-idf//components/esp_event/include', '-I/data_pool/micropython/esp/esp-idf//components/tcpip_adapter/include', '-I/data_pool/micropython/esp/esp-idf//components/lwip/lwip/src/include', '-I/data_pool/micropython/esp/esp-idf//components/lwip/port/esp32/include', '-I/data_pool/micropython/esp/esp-idf//components/lwip/include/apps', '-I/data_pool/micropython/esp/esp-idf//components/lwip/include/apps/sntp', '-I/data_pool/micropython/esp/esp-idf//components/mbedtls/mbedtls/include', '-I/data_pool/micropython/esp/esp-idf//components/mbedtls/port/include', '-I/data_pool/micropython/esp/esp-idf//components/mdns/include', '-I/data_pool/micropython/esp/esp-idf//components/mdns/private_include', '-I/data_pool/micropython/esp/esp-idf//components/spi_flash/include', '-I/data_pool/micropython/esp/esp-idf//components/ulp/include', '-I/data_pool/micropython/esp/esp-idf//components/vfs/include', '-I/data_pool/micropython/esp/esp-idf//components/xtensa-debug-module/include', '-I/data_pool/micropython/esp/esp-idf//components/wpa_supplicant/include', '-I/data_pool/micropython/esp/esp-idf//components/wpa_supplicant/port/include', '-I/data_pool/mic [... cut ...] imble/porting/npl/freertos/include', '-I/data_pool/micropython/esp/esp-idf//components/bt/host/nimble/nimble/ext/tinycrypt/include', '-I/data_pool/micropython/esp/esp-idf//components/bt/host/nimble/esp-hci/include', '-I../../examples/usercmodule//cppexample', '-DNO_QSTR', '../../examples/usercmodule/cppexample/example.cpp']' returned non-zero exit status 1.
make: *** [../../py/mkrules.mk:105: build-GENERIC/genhdr/qstr.i.last] Error 1
make: *** Deleting file 'build-GENERIC/genhdr/qstr.i.last'
```

wgaylord, 2020-12-19:
Someone on the forums pointed out to try adding
`CXXFLAGS_USERMOD += -I$(BOARD_DIR) -Wno-error=cpp -Wno-error=sign-compare`

to the micropython.mk file in the cppexample folder. This fixes the issue.

dpgeorge, 2021-04-01:
This should now work (the esp32 switched to use CMake).

closed 2021-04-01 by dpgeorge

wgaylord, 2021-04-01:
Actually building user modules don't work at all right now from what I can tell.

dpgeorge, 2021-04-02:
> Actually building user modules don't work at all right now from what I can tell.

Can you please explain how you are building them and what goes wrong?  The CI builds a few ports with user C modules, so it should be working.

wgaylord, 2021-04-02:
I build using the make command as instructed in the ESP32 readme. And the problem is when I enable them the build just fails to get past the generating the secondary make file stage.

dpgeorge, 2021-04-02:
> And the problem is when I enable them the build just fails to get past the generating the secondary make file stage.

Can you please post the exact command you used, the IDF version, and the full output?