micropython/micropython #6710: Wifi DNS setting is changing all the time on ESP-32

TKFRvisionOfficial, 2020-12-19:
I have a ESP-32 with MicroPython v1.13.
Everytime I set the ifconfig with ("192.168.2.160", "255.255.255.0", "192.168.2.160", "8.8.8.8") it switches the dns (8.8.8.8) back to 254.128.0.0 and I have no access to the internet anymore.
Thanks

stonecharioteer, 2020-12-21:
That is really strange. `254.128.0.0` is not hardcoded into this repository. Do you have instructions for what you've done so I can test it?

jwshields, 2020-12-31:
I was experiencing the same issue when I was running uPy built on idf3.
I posted about it in https://github.com/micropython/micropython/issues/6495
When I flashed the idf4 build, the problem went away. You may want to give that a try.. But that doesn't necessarily solve the underlying issue of the idf3 version having DNS be all wonky..

TKFRvisionOfficial, 2020-12-31:
But won't IDF4 remove some features?

jwshields, 2021-01-02:
Mind, I am still learning more and more about uPy, so I don't want to appear to be an authoritative source of information..
But, based on the [download page](https://micropython.org/download/esp32/)

> Firmware built with ESP-IDF v3.x, with support for BLE, LAN and PPP
> Firmware built with ESP-IDF v4.x, with support for BLE, but no LAN or PPP

If these are the "only" or "main" differences, I am personally okay without these features: LAN and PPP. 
I mentioned my own issue above, 6495 - and the first commenter seemed to be aghast that I was using the firmware built with IDF3.

As a newcomer to uPy and ESP chips.... The details of why/what/how things are different are lost to me, and a hand wavy "use 3 rather than 4" suggestion feels very misleading because of issues like this (ie- "dns settings not sticking")

tve, 2021-01-02:
I'd totally agree. I have a dozen esp32's out in the field running different projects and most are running esp-idf 4.0.1, the newer ones are now on 4.1, and I wish I could get the cmake builds to work with 4.2. I've not had any use for LAN or PPP so the thought of using 3.x wouldn't cross my mind, given how many improvements esp-idf 4.x has. However, the powers that be don't seem to use esp32's on a regular basis and are not interested in updating the recommendation. Oh well...

jonnor, 2024-09-13:
We are now on esp-idf 5.x, which should render this fixed.

closed 2024-09-13 by jonnor