micropython/micropython #6712: Micropython returns native word sized ints from almost all functions.

133794m3r, 2020-12-20:
Micropython is more than happy to let `2**128` return a python int object but for `randint(2**33,2**33-1)` Micropython will attempt to clamp that range down to a 32bit number and warn of an overflow if the value is more than 32bits in size. This behavior is not inline with the rest of the arithmetic options of the environment. For some reason the randint function and the getrandbits function both attempt to clamp it down to a 32bit value.

This makes no sense as I can very easily have very large ints elsewhere in the functions. This kind of "half-in" "half-out" behavior makes no sense. It's also no listed _anywhere_ in the documentation about this strange behavior. The `pow` function just operating on floating point numbers is fine as there's an easy workaround within the language itself but randint trying to clamp down to a machine word instead of just returning an int object is beyond me.

jimmo, 2020-12-22:
I think probably the main reason is that handling this case would make this code significantly more complicated (and therefore take up more ROM). The workaround for this (get 31 bits of randomness at a time and shift+and into a result) can be done in Python relatively easily.

This could be added as a "Difference to CPython" though. The way to do that is to add a test to `tests/cpydiff`.

There is a bug here on 64-bit architectures though, see https://github.com/micropython/micropython/issues/6718

133794m3r, 2020-12-22:
getrandbits() will just return a ValueError(null)  which doesn't help at all with the end user figuring out what's going on of you try to get 32bits of random data. It should also just throw an error that you're trying to get more than 31bits of state.

peterhinch, 2020-12-22:
Three MicroPython pseudorandom number generators may be found here: [this one](https://github.com/peterhinch/micropython-samples/blob/master/random/random.py) can produce up to 64 bit results. Strictly for amusement purposes only: don't use for encryption or other security application.

133794m3r, 2020-12-22:
I'm using it for demonstrations for textbook RSA. The point is for it to be broken. I only really care about getting a number that's the size of bits I need and then finding the next prime after it(I've already got code that works well enough for that.) I just need a prime of the size specified and that's it. It's for CTF challenges.

dpgeorge, 2021-05-30:
The docs were updated in 34d4dab683c69578b0ddd1725290529dd812b291

It's possible to use the `random.py` module from `micropython-lib` to overcome these issues with `randint()` and `getrandbit()`.

closed 2021-05-30 by dpgeorge