micropython/micropython #6714: extmod/uasyncio/stream.py: Close a socket conection

IhorNehrutsa, 2020-12-21:
I'm not sure here.
1)It seems that the connection may be closed with not empty Stream.out_buf
2)It seems that the connection may be closed with not fully transferred data in the queue inside the poller
https://github.com/micropython/micropython/blob/069557edef84a4f37ad52ae93838129da8e5fe77/extmod/uasyncio/stream.py#L27

IhorNehrutsa, 2021-12-28:
@peterhinch 
Could you pay attention?

peterhinch, 2021-12-28:
This is a question for the maintainers. The real issue is whether MicroPython conforms with CPython behaviour: if you can identify a difference then there is an issue.

As a general point, closing an I/O device is drastic. You wouldn't close a file without first ensuring that I/O to it was complete.

closed 2023-03-12 by IhorNehrutsa