micropython/micropython #6719: namedtuple crash with empty field list

jimmo, 2020-12-22:
On the Unix port:

```
>>> from collections import namedtuple
>>> n = namedtuple('a', '')
>>> n()
Segmentation fault (core dumped)
```

The crash comes from `namedtuple_make_new` which tries to set `tuple->base.type = type_in;`, but `tuple` is `mp_const_empty_tuple`.

The crash didn't happen on PYBD, however it appears to be overwriting the type of mp_const_empty_tuple.

Additionally, MicroPython allows a namedtuple with an empty name, e.g.

```
>>> namedtuple('', 'a')
<class ''>
```

whereas CPython:

```
>>> namedtuple('', 'a')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.8/collections/__init__.py", line 358, in namedtuple
    raise ValueError('Type names and field names must be valid '
ValueError: Type names and field names must be valid identifiers: ''
```

stinos, 2020-12-22:
> MicroPython allows a namedtuple with an empty name,

Can be made into a known difference with CPython, no? Shouldn't be a real problem, unlike the empty field list handling.

dpgeorge, 2022-07-19:
> Additionally, MicroPython allows a namedtuple with an empty name

Let's leave that as a hidden corner of MicroPython.  It does no harm.

closed 2022-07-18 by dpgeorge