micropython/micropython #6720: RFC ESP32 application performance relative to STM

peterhinch, 2020-12-22:
tl;dr There is evidence that application performance of ESP32 is substantially below that of the Pyboard 1.1.

This was prompted by two observations. In [this issue](https://github.com/peterhinch/micropython-async/issues/52) users noted that on ESP32 the GPS demo failed at message rates above 2Hz, whereas on a Pyboard 1.1 it was tested at 10Hz. Secondly [lcd160cr-gui](https://github.com/peterhinch/micropython-lcd160cr-gui) produces snappy screen changes on a Pyboard, but is visibly sluggish on an ESP32.

I wrote a script (listing below) which launches a `metrics` task which does nothing except maintain statistics of how frequently it is scheduled. Every 10s it issues a report to the REPL which includes these figures:
 * Maximum, minimum and average interval between the task being scheduled.
 * Mean frequency with which the task is scheduled.
 * Free RAM.

[This demo script](https://github.com/peterhinch/micropython-lcd160cr-gui/blob/master/gui/demos/lbt.py) was modified to launch the `metrics` task. While the demo ran, I operated the touch GUI. The reports with the worst value of maximum interval are reproduced below:
```
Pyboard 1.1
Max 72148μs Min 273μs Avg 1955μs No. of calls 5113 Freq 511
mem free 34928
ESP32
Max 1460920μs Min 1102μs Avg 6363μs No. of calls 1571 Freq 157
mem free 43152
```
The worst case on ESP32 is ~20x slower, the mean scheduling rate (measured by average time or frequency) is consistently about 3x slower on ESP32. In both cases over many measurements free RAM was similar, varying from ~35-47K as I operated the GUI with the ESP32 being generally rather better than the Pyboard.

I would be interested to hear any views on why this is the case. Neither the GUI nor the GPS applications use interrupts or timers: they are pure **uasyncio V3** code using polling (touch GUI) and stream I/O (GPS). Apart from hardware setup, code is identical on the two platforms.

Running **metrics.py** as a stand-alone benchmark produced this outcome (very consistent between reports):
```
ESP32
Max 5489μs Min 2061μs Avg 2232μs No. of calls 4479 Freq 447
mem free 106528
Pyboard 1.1
Max 480μs Min 449μs Avg 478μs No. of calls 20901 Freq 2090
mem free 98192
```
**metrics.py**
```python
import uasyncio as asyncio
import gc
from utime import ticks_us, ticks_diff

def metrics():
    ncalls = 0
    max_d = 0
    min_d = 100_000_000
    tot_d = 0
    st = 'Max {}μs Min {}μs Avg {}μs No. of calls {} Freq {}'
    async def func():
        nonlocal ncalls, max_d, min_d, tot_d
        while True:
            tstart = ticks_us()
            t_last = None
            while ticks_diff(t := ticks_us(), tstart) < 10_000_000:
                await asyncio.sleep(0)
                if ncalls:
                    dt = ticks_diff(t, t_last)
                    max_d = max(max_d, dt)
                    min_d = min(min_d, dt)
                    tot_d += dt
                ncalls += 1
                t_last = t
            print(st.format(max_d, min_d, tot_d//ncalls, ncalls, ncalls//10))
            gc.collect()
            print('mem free', gc.mem_free())
            ncalls = 0
            max_d = 0
            min_d = 100_000_000
            tot_d = 0
    return func

# Example of call
async def main():
    asyncio.create_task(metrics()())  # Note the syntax
    while True:
        await asyncio.sleep(0)

# asyncio.run(main())
```

peterhinch, 2020-12-22:
I suppose I should clarify what comments I'm hoping for:

* Is this level of performance in line with expectations?
* Is it worth my investigating further? If so, ideas on how to do so would be welcome.
* What on earth is my ESP32 **doing** for 1.4s? Is it likely that FreeRTOS is hogging the CPU?

tve, 2020-12-22:
A couple of comments about performance tuning of MP on esp32:
- you should compile your own firmware with the clock rate set to 240Mhz: `SDKCONFIG += boards/sdkconfig.240mhz` in mpconfigport.mk
- your should also force flash 80Mhz QIO access and ensure some ancient works-arounds are disabled in sdkconfig.board (you can see the chip version with esptool or in the boot messages):
```
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESP32_REV_MIN_1=y
```
- you can see what the boot process configures using `CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y # useful to see how flash/psram is inited` in sdkconfig.board
- most (any?) stream I/O activity occurs using polling and the interval is 10ms minimum, which can cause quite some delays, especially with small packets, you should change the context switch rate in freertos to 1000Hz (which is widely use and perhaps even the default in FreeRTOS, just not in esp-idf and MP):
```
CONFIG_FREERTOS_USE_TICKLESS_IDLE=y
CONFIG_FREERTOS_HZ=1000
```
- unless you need bluetooth you should also disable bluetooth support 'cause it disables one of the esp32 cores (don't ask me what I think about this): `MICROPY_PY_BLUETOOTH = 0` in mkconfigboard.mk

I'm not saying this will miraculously fix all your performance issues, but unless you do this it's kind'a pointless to compare against pyboard... It baffles me that the powers that be want to keep the std config so pathetic but I've stopped trying to understand.

peterhinch, 2020-12-23:
Thank you, that is very useful information especially re I/O: this may have a lot of bearing on the GPS application. As you probably know, `uasyncio` stream I/O does indeed use the poll mechanism.

I aim to make my applications and demos work with official firmware, but with this information at least users have a choice.

UnexpectedMaker, 2021-01-11:
@peterhinch I'm not sure this has any bearing, but 1.13 on ESP32 got some significant slowdowns since the BLE CPU pinning change that happened after 1.13 came out:
https://github.com/micropython/micropython/issues/6565

jonnor, 2024-09-13:
Are there still issues to address here?

peterhinch, 2024-09-14:
I think this can be considered fixed.

closed 2024-09-14 by peterhinch