micropython/micropython #6723: CC3200 debug build is too large for WiPy

VinDuv, 2020-12-23:
I’m attempting to create a debug build of the CC3200 WiPy port to run on my WiPy v1.3.
(Note that passing `BTYPE=debug` to the Makefile does not currently produce a debug build, since the Makefile fails to set `-DDEBUG`. I’ve submitted PR #6722 to fix this issue).
The problem is that the debug build is substantially larger than the release build, 205K vs 182K. This exceeds the maximum size of 192K defined in [ports/cc3200/bootmgr/flc.h#L59](https://github.com/micropython/micropython/blob/master/ports/cc3200/bootmgr/flc.h#L59).
Would it be possible to change this size so the debug build could be used?

davehylands, 2020-12-23:
That device only has 256K total RAM. Bascially the way its divided up now is that 192K is for firmware, 16K is for the bootloader and 48K is for heap. Whatever amount you increase for the firmware, you need to decrease the heap by the same amount.

You'd be better off to only build the actual code that you want to debug with debugging enabled, unless you can live with the smaller heap.

jimmo, 2020-12-29:
@VinDuv it's also worth noting that the BTYPE flag only sets the DEBUG/NDEBUG macro, it still leaves optimisations (`-Os`) enabled, so debugging isn't going to be very useful. Like @davehylands says, your best bet is to leave DEBUG unset, and just selectively add debugging code where you need (or set `-Og` instead, and use an interactive debugger rather than printf).

jonnor, 2024-08-18:
There are solutions proposed above for selective debugging. Full debugging everywhere is difficult for such constrained targets.

Proposing to close this issue.

robert-hh, 2024-08-18:
Additionally the CC3200 port is stale and subject to be closed latest with v2.0.

closed 2024-08-18 by robert-hh