micropython/micropython #6725: Blockdevice2 not working with littleFS

hoihu, 2020-12-23:
We have a somewhat similar HW setup as on the pyboard D-series:

- STM32F7
- one spi bank is connected via softspi bitbang
- the other spi bank is connected via HW SPI
- both are 128MBits in size

the resulting blockdevices were used to create an FS on /flash (spi1) and /data (spi2). That's a difference to the pyboard D, where the HW QSPI is used for XPI.

/flash was formatted as littleFS /data as FAT. everything worked so far.

We wanted to reformat spi2 to littleFS too, but strangely it reports a "OSError(28)". 

Consider this function:

```
def format_drive(name, bdev):
    try:
        os.umount(name)
    except OSError:
        print("could not properly unmount {}".format(name))
        
    print("formatting {}...".format(name))
    os.VfsLfs2.mkfs(bdev)
    os.mount(bdev, name)

format_drive('/flash', pyb.Flash(start=0, len=16777216))
format_drive('/data', pyb.Flash(start=16777216, len=16777216))
print("done")   
```

the resulting output was

```
formatting /flash...
could not properly unmount /data
formatting /data...
Traceback (most recent call last):
  File "<stdin>", line 15, in <module>
  File "<stdin>", line 11, in format_drive
OSError: 28
```

so basically this works:
```os.VfsFat.mkfs(pyb.Flash(start=16777216, len=16777216))```

whereas this doesn't:
```os.VfsLfs2.mkfs(pyb.Flash(start=16777216, len=16777216))```

Any ideas or pointers where to look at? Could be thrown here, but I'm not into the details
https://github.com/micropython/micropython/blob/master/extmod/vfs_lfsx.c#L145 

as background info: this is the Macro we use for the block device 2:
```
// SPI flash #2, block device config
extern const struct _mp_spiflash_config_t spiflash2_config;
extern struct _spi_bdev_t spi_bdev2;

#define MICROPY_HW_BDEV2_IOCTL(op, arg) ( \
    (op) == BDEV_IOCTL_NUM_BLOCKS ? ((1UL << (MICROPY_HW_QSPIFLASH_SIZE_BITS_LOG2 - 3)) / FLASH_BLOCK_SIZE) : \
    (op) == BDEV_IOCTL_INIT ? spi_bdev_ioctl(&spi_bdev2, (op), (uint32_t)&spiflash2_config) : \
    spi_bdev_ioctl(&spi_bdev2, (op), (arg)) \
)
#define MICROPY_HW_BDEV2_READBLOCKS(dest, bl, n) spi_bdev_readblocks(&spi_bdev2, (dest), (bl), (n))
#define MICROPY_HW_BDEV2_WRITEBLOCKS(src, bl, n) spi_bdev_writeblocks(&spi_bdev2, (src), (bl), (n))
```

jonnor, 2024-08-18:
Hi @hoihu - thank you for reporting. Did you find a solution for this? Is the issue still present in latest MicroPython (1.23)?

projectgus, 2024-09-10:
I'm going to close this, as I wouldn't be surprised if the issue still exists but hard to test without the specific hardware. @hoihu if you get back to this at some point and the problem is still there then please ping me, we can reopen and try to dig into it.

closed 2024-09-10 by projectgus