micropython/micropython #6737: SSL Handshake error

rdehuyss, 2020-12-27:
Hi,

I'm the author of [micropython-ota-updater](https://github.com/rdehuyss/micropython-ota-updater).

For some reason, I get an SSL handshake error in an ESP8266 board and not on an ESP32 board.

Steps to reproduce:

Create the following file as `httpclient.py`:

```python
import usocket, os

class Response:

    def __init__(self, socket, saveToFile=None):
        self._socket = socket
        self._saveToFile = saveToFile
        self._encoding = 'utf-8'
        if saveToFile is not None:
            CHUNK_SIZE = 512 # bytes
            with open(saveToFile, 'w') as outfile:
                data = self._socket.read(CHUNK_SIZE)
                while data:
                    outfile.write(data)
                    data = self._socket.read(CHUNK_SIZE)
                outfile.close()
                
            self.close()

    def close(self):
        if self._socket:
            self._socket.close()
            self._socket = None

    @property
    def content(self):
        if self._saveToFile is not None:
            raise SystemError('You cannot get the content from the response as you decided to save it in {}'.format(self._saveToFile))

        result = None
        try:
            result = self._socket.read()
        finally:
            self.close()
        return result

    @property
    def text(self):
        return str(self.content, self._encoding)

    def json(self):
        import ujson
        return ujson.loads(self.content)

class HttpClient:

    def __init__(self, headers={}):
        self._headers = headers

    def request(self, method, url, data=None, json=None, file=None, custom=None, saveToFile=None, headers={}, stream=None):
        def _write_headers(sock, _headers):
            for k in _headers:
                sock.write(b'{}: {}\r\n'.format(k, _headers[k]))

        try:
            proto, dummy, host, path = url.split('/', 3)
        except ValueError:
            proto, dummy, host = url.split('/', 2)
            path = ''
        if proto == 'http:':
            port = 80
        elif proto == 'https:':
            import ussl
            port = 443
        else:
            raise ValueError('Unsupported protocol: ' + proto)

        if ':' in host:
            host, port = host.split(':', 1)
            port = int(port)

        ai = usocket.getaddrinfo(host, port, 0, usocket.SOCK_STREAM)
        if len(ai) < 1:
            raise ValueError('You are not connected to the internet...')
        ai = ai[0]

        print(ai)

        s = usocket.socket(ai[0], ai[1], ai[2])
        try:
            s.connect(ai[-1])
            if proto == 'https:':
                s = ussl.wrap_socket(s, server_hostname=host)
            s.write(b'%s /%s HTTP/1.0\r\n' % (method, path))
            if not 'Host' in headers:
                s.write(b'Host: %s\r\n' % host)
            # Iterate over keys to avoid tuple alloc
            _write_headers(s, self._headers)
            _write_headers(s, headers)

            # add user agent
            s.write(b'User-Agent: MicroPython Client\r\n')
            if json is not None:
                assert data is None
                import ujson
                data = ujson.dumps(json)
                s.write(b'Content-Type: application/json\r\n')

            if data:
                s.write(b'Content-Length: %d\r\n' % len(data))
                s.write(b'\r\n')
                s.write(data)
            elif file:
                s.write(b'Content-Length: %d\r\n' % os.stat(file)[6])
                s.write(b'\r\n')
                with open(file, 'r') as file_object:
                    for line in file_object:
                        s.write(line + '\n')
            elif custom:
                custom(s)
            else:
                s.write(b'\r\n')

            l = s.readline()
            # print(l)
            l = l.split(None, 2)
            status = int(l[1])
            reason = ''
            if len(l) > 2:
                reason = l[2].rstrip()
            while True:
                l = s.readline()
                if not l or l == b'\r\n':
                    break
                # print(l)
                if l.startswith(b'Transfer-Encoding:'):
                    if b'chunked' in l:
                        raise ValueError('Unsupported ' + l)
                elif l.startswith(b'Location:') and not 200 <= status <= 299:
                    raise NotImplementedError('Redirects not yet supported')
        except OSError:
            s.close()
            raise

        resp = Response(s, saveToFile)
        resp.status_code = status
        resp.reason = reason
        return resp

    def head(self, url, **kw):
        return self.request('HEAD', url, **kw)

    def get(self, url, **kw):
        return self.request('GET', url, **kw)

    def post(self, url, **kw):
        return self.request('POST', url, **kw)

    def put(self, url, **kw):
        return self.request('PUT', url, **kw)

    def patch(self, url, **kw):
        return self.request('PATCH', url, **kw)

    def delete(self, url, **kw):
        return self.request('DELETE', url, **kw)
```

In the repl, run the following command:
```
from httpclient import HttpClient
h = HttpClient()
h.get('https://github.com/rdehuyss/micropython-ota-updater/blob/master/__init__.py') # this works!
h.get('https://raw.githubusercontent.com/rdehuyss/micropython-ota-updater/0.5.3/app/__init__.py') # this does not work?
```

I analyzed the SSL certificates and they are almost identical.

Any idea what is going on?

kevinjwalters, 2021-01-29:
What error do you get? Does the order of those two `get()` calls make any difference?

kevinjwalters, 2021-01-29:
One observation: `raw.githubusercontent.com:443` is spitting a [TLS session ticket](https://en.wikipedia.org/wiki/Transport_Layer_Security#Resumed_TLS_handshake) whereas `github.com:443` does not.

Point 5 in http://docs.micropython.org/en/latest/esp8266/general.html?highlight=tls#ssl-tls-limitations could be relevant too.

jonnor, 2024-09-02:
This is very likely due to the above mentioned limitations in SSL support on ESP8266, which are unfortunately not really fixable. Proposing to close this issue.

closed 2024-09-02 by dpgeorge