micropython/micropython #6741: lfs2: crash in lfs2_pair_cmp

tve, 2020-12-29:
I'm getting sometimes persistent crashes on an esp32 running master post v1.13. It happens when closing a just-written file. I'm not sure what to do about it yet, posting here in case others bump into the same thing.
The backtrace is:
```
PC: 0x401c1a7f: lfs2_pair_cmp at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../lib/littlefs/lfs2.c:267
BT-0: 0x401c1a7c: lfs2_pair_cmp at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../lib/littlefs/lfs2.c:266
BT-1: 0x400ff4c2: lfs2_dir_commit at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../lib/littlefs/lfs2.c:1829 (discriminator 1)
BT-2: 0x400fff69: lfs2_file_rawsync at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../lib/littlefs/lfs2.c:2841
BT-3: 0x400fff91: lfs2_file_rawclose at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../lib/littlefs/lfs2.c:2622
BT-4: 0x401007c1: lfs2_file_close at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../lib/littlefs/lfs2.c:5091
BT-5: 0x400f5e40: mp_vfs_lfs2_file_ioctl at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../extmod/vfs_lfsx_file.c:190
BT-6: 0x400e9add: mp_stream_close at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../py/stream.c:422
BT-7: 0x400e387a: fun_builtin_1_call at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../py/objfun.c:71
BT-8: 0x400df8cd: mp_call_function_n_kw at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../py/runtime.c:650
BT-9: 0x400df9fa: mp_call_method_n_kw at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../py/runtime.c:666
BT-10: 0x400ed4e9: mp_execute_bytecode at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../py/vm.c:1085
```
The exception is `(LoadProhibited). Exception was unhandled.` and the exception address is 0x17f8

tve, 2020-12-29:
Another lfs2 crash:
```
PC: 0x400fecbc: lfs2_fs_rawtraverse at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../lib/li
ttlefs/lfs2.c:3849
BT-0: 0x400fecb9: lfs2_fs_rawtraverse at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../lib/
littlefs/lfs2.c:3848
BT-1: 0x400fede1: lfs2_alloc at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../lib/littlefs/
lfs2.c:572
BT-2: 0x400fee95: lfs2_file_relocate at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../lib/l
ittlefs/lfs2.c:2644
BT-3: 0x400ff03a: lfs2_file_outline at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../lib/li
ttlefs/lfs2.c:2714
 (inlined by) lfs2_file_rawwrite at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../lib/littl
efs/lfs2.c:2973
BT-4: 0x40100806: lfs2_file_write at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../lib/litt
lefs/lfs2.c:5143
BT-5: 0x400f5eb5: mp_vfs_lfs2_file_write at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../e
xtmod/vfs_lfsx_file.c:150
BT-6: 0x401c0f85: mp_stream_rw at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../py/stream.c
:62
BT-7: 0x400e9ee5: mp_stream_write at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../py/strea
m.c:230
BT-8: 0x400e9f4e: stream_write_method at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../py/s
tream.c:264
BT-9: 0x400e394e: fun_builtin_var_call at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../py/
objfun.c:131
BT-10: 0x400df8cd: mp_call_function_n_kw at /home/src/esp32/mpy-1.13/micropython/ports/esp32/../../p
y/runtime.c:650
```
`Core  0 panic'ed (LoadProhibited). Exception was unhandled. -- EXCVADDR: 0x05676f72`
Smells like something get overwritten in the file handling code...

straga, 2021-03-05:
Probably same: https://github.com/micropython/micropython/issues/6568