micropython/micropython #6747: Wifi credentials are not store to flash

ondratu, 2021-01-01:
Hi,

this is important different wifi think between ESP8266 and ESP32. ESP8266 store wifi credentials by default with ESP API, but for ESP32 you need to call `esp_wifi_set_storage` with right value, end call `esp_wifi_set_auto connect`, which is not call in modnetwork.c en ESP32 port.

So, it will be fine, when this additional API for ESP32 will be wrapped to python library. Many flash area overwriting crash the flash, but without storing credentials to flash devices will be more vulnerable. 

And **mainly**, this can be documented well in examples, because, code for ESP8266, which do auto-connect with stored credentials not work on ESP32!

peterhinch, 2021-01-01:
This is not a bug. It's a deliberate design difference between the two hardware platforms. It is a FAQ which has been discussed at length in [the forum](http://forum.micropython.org/index.php).

ondratu, 2021-01-02:
It is ok, but i think that the low ESP API can be wrapped to python. But this difference can be somewhere in esp32 port documentation. Why? I was try to google information, and only what i get was information from C API, credentials are stored to flash too as on ESP8266. When some important information is hidden in source code, that is bad!

And in my opinion, when can be do with same functionality, i can be do that by default. Why? For easier platform switch. But my issue is not about that.

jonnor, 2024-08-18:
It is not planned to implement ESP32 specific storing of WiFi credentials.
The recommended solutions are to use an external WiFi credential manager is advised, or manage the storage yourself with standard MicroPython functionality such as file read/write.

Proposing to close this issue.

closed 2024-08-18 by robert-hh