micropython/micropython #6749: .mpy on Micropython 1.13 not working

lutecky, 2021-01-03:
Hi,
I convert the .py files to .mpy on macOS. After uploading to ESP8266, importing these files does not work.

**mpy-cross version:**
Kuba-MacBook-Pro:Software ver 2 kuba$ mpy-cross --version
MicroPython v1.13 on 2021-01-01; mpy-cross emitting mpy v5

**Micropython on ESP8266:**
MicroPython v1.13 on 2020-09-11; ESP module with ESP8266

I try use mpy-cross like:
mpy-cross -mcache-lookup-bc ./myfile.py
mpy-cross ./myfile.py
mpy-cross -march=xtensa ./myfile.py

lutecky, 2021-01-06:
>>> import mymodule
Trackback (most recent call last):
     File "<stdin>", line 1, in <module>
ValueError: invalid syntax for number

robert-hh, 2021-01-06:
mpy-cross -march=xtensa inputfile
lshouid work. What is the output of calling mpy-cross --version?

lutecky, 2021-01-06:
Hi,
Now I saw that compilation for some modules it works. There is a problem with two .py files, but I don't know why I have that issue - original py file works good, after compilation during import I see a message:

_Trackback (most recent call last):
File "", line 1, in 
ValueError: invalid syntax for number_

robert-hh, 2021-01-06:
Are you sure that you uploaded the newly compiled files to the target device, and that there is no frozen module with the same name?
I can compile and import that file w/o error messages.

lutecky, 2021-01-06:
Yes, I'm sure on 100%. Can you send to me your mpy file of that module: https://pastebin.com/98h7w2ew
During compilation there are no errors, when I try to import the module for the first time the above mentioned error is displayed, after the second approach it is no longer displayed but the module will not be imported.

[image]

robert-hh, 2021-01-06:
OK. Here it is:
[myfile.zip](https://github.com/micropython/micropython/files/5776683/myfile.zip)

lutecky, 2021-01-06:
Robert, your .mpy file are working. I don't know why but my .mpy have 1205 bytes and yours 1288 bytes...

That's my
[mcp.mpy.zip](https://github.com/micropython/micropython/files/5776770/mcp.mpy.zip)

robert-hh, 2021-01-06:
You can find my versions of of mpy-cross at https://github.com/robert-hh/Shared-Stuff. Files:

mpy-cross_micropython_org_linux
mpy-cross.exe

lutecky, 2021-01-06:
Robert, I try use your version of mpy-cross but you threw a linux binary file and I used macOS. If you can share the sources, I'll be happy to compile and send you a version for macOS.

robert-hh, 2021-01-06:
The sources are here: https://github.com/micropython/micropython
subdirectory mpy-cross

lutecky, 2021-01-06:
I use this version, I have these problems with it. I thought your version was "improved".

robert-hh, 2021-01-06:
I compiled the Linux and windows versions from these sources. So it may be a problem with the build process of mpy-cross on Mac.

lutecky, 2021-01-06:
OK, thanks for your support! I will try to fix that bug. Could you convert one library for me, because I am standing still now?
[tinyweb.py.zip](https://github.com/micropython/micropython/files/5777219/tinyweb.py.zip)

robert-hh, 2021-01-06:
Yes. Here it is. 
[tinyweb.zip](https://github.com/micropython/micropython/files/5777349/tinyweb.zip)
Again: what is the output of running the command:
`mpy-cross --version`
It should be something like:
`MicroPython v1.13-271-g9d07b2a7a on 2021-01-06; mpy-cross emitting mpy v5`
The part "emitting mpy v5" is important.

lutecky, 2021-01-06:
>Kuba-MacBook-Pro:Software ver 2 kuba$ mpy-cross --version
MicroPython v1.13 on 2021-01-01; mpy-cross emitting mpy v5

robert-hh, 2021-01-06:
I just get the error : `ImportError: no module named 'logging'`
So it looks OK. The tested binary is below.
[tinyweb.zip](https://github.com/micropython/micropython/files/5777536/tinyweb.zip)
You could try to load your sources into an editor and store them again. Maybe it's a line ending MAC vs. Linux issue.

lutecky, 2021-01-06:
I don't know why, but it not working. I changed the firmware to **esp8266-20200911-v1.13.bin** but issue still persist.

robert-hh, 2021-01-07:
Which board are you using?

lutecky, 2021-01-07:
I used ESP8266-12 with 4MB memory

robert-hh, 2021-01-07:
That's strange. I tested it here again with an ESP8266-12 with 4MB flash, and an ES8266-01, pimped with 16 MB flash, and all work. The firmware I made for that is https://github.com/robert-hh/Shared-Stuff/blob/master/firmware-ESP8266.bin
It has a few additional frozen scripts, like a ftp sever, and editor and upysh.py from micropython.lib. But that does not affect basic operation.

lutecky, 2021-01-10:
The cause of the problem (with tinyweb) is logging module. After comment all "log" the module has started to work!

```
#import logging
#log = logging.getLogger('WEB')

#log.(...)
```

robert-hh, 2021-01-10:
That looks like a free memory problem. I had no logging module on my device, so I got an import error. After copying in the logging module from micropython-lib, I get an memory error. That's because during importing the memory is exhausted. It works if I also pre-compile logging.py and handlers.py. But after that the free heap is substantially smaller. Like 17k free remaining.

closed 2022-03-30 by dpgeorge