micropython/micropython #6750: Tests failed with Python 3.9.1

yangfl, 2021-01-04:
https://github.com/micropython/micropython/blob/f7aafc0628f2008d015b32b0c0253a13f748d436/tests/extmod/urandom_basic.py#L25-L29

Python now accepts `random.getrandbits(0)`, it simply returns 0.

https://github.com/micropython/micropython/blob/f7aafc0628f2008d015b32b0c0253a13f748d436/tests/import/pkg7/subpkg1/subpkg2/mod3.py#L7-L11

Python now throws `ImportError` instead `ValueError`

```
Traceback (most recent call last):
  File "/.../micropython/tests/import/import_pkg7.py", line 2, in <module>
    import pkg7.subpkg1.subpkg2.mod3
  File "/.../micropython/tests/import/pkg7/subpkg1/subpkg2/mod3.py", line 9, in <module>
    from .... import mod1
ImportError: attempted relative import beyond top-level package
```

dpgeorge, 2021-05-30:
> Python now throws `ImportError` instead `ValueError`

Do you know if there's a CPython PEP or bpo for this change?

dpgeorge, 2021-05-30:
`getrandbits(0)` is fixed in c3199f56494429ba44f7e722597945dc91f3589c

dpgeorge, 2021-05-30:
> > Python now throws `ImportError` instead `ValueError`
> 
> Do you know if there's a CPython PEP or bpo for this change?

It's here: https://bugs.python.org/issue37444

dpgeorge, 2021-05-30:
See #7336 for a fix for the relative import error.

dpgeorge, 2021-05-30:
Relative import error fixed in 53519e322a5a0bb395676cdaa132f5e82de22909

PR #7336 (2021-05-30): py/builtinimport: Change relative import's ValueError to ImportError.
Following CPython change, see https://bugs.python.org/issue37444.

Fixes #6750.
closed 2021-05-30 by dpgeorge