micropython/micropython #6752: [ESP32] Access to clock gating registers from .mpy modules

Winkelkatze, 2021-01-05:
Hi everyone,

I wanted to implement a hardware driver as .mpy module. I know that these modules don't have access to the IDF functions but since my code is using its own peripheral driver anyway, I wanted to build it as a module to be more convenient to use and share. GPIO configuration can be done through the 'gpio_matrix_out' functions exposed in the 'esp' module. But the only problem is, that I can't access the peripheral clock gating registers safely.

I wonder what you guys are thinking about exposing the following functions
- periph_module_enable
- periph_module_disable
- periph_module reset
through the 'esp32' module?

Or would it be better to make platform-specific functions available to dynamically loaded modules?

Or is my plan to build a hardware driver as a native .mpy module just a bad idea? :)

jimmo, 2021-01-06:
Hi @Winkelkatze ,

> Or would it be better to make platform-specific functions available to dynamically loaded modules?

This is a bit of an open problem at the moment. https://github.com/micropython/micropython/pull/5711 (and linked PRs) probably contains the most information.

> Or is my plan to build a hardware driver as a native .mpy module just a bad idea?

This works moderately well on other architectures where a driver just needs to read/write registers at known memory locations, but ESP32 makes this a lot harder.

I think the approach in #5711 (or something like it) is probably worth considering. At least on ESP32 the additional code size isn't so much of a problem. I definitely like the idea that this allows for a much wider ecosystem of ESP32 drivers provided by third-party modules.

> ... about exposing the following functions ... through the 'esp32' module?

I guess this is useful if you're planning to access them from Python, but sounds like generally you want this from native code? Is there a need for these from Python?

Winkelkatze, 2021-01-06:
Ahh damn, should have spent more time looking through the PRs :) Thanks @jimmo, the one you linked exactly addresses that issue.

Accessing these functions from python in general would be not be that useful. The only potential use I can see is power management, but this should better be done through the respective drivers for the hardware.

So I guess, I'll just wait for new developments in #5711.

BTW implementing hardware drivers for the ESP controllers isn't really different from other controllers, stuff is still at fixed memory locations. And everything (except WiFi and BT) is reasonably well documented. The dual core structure can make things a bit more tricky though.