micropython/micropython #6759: Difference between USB_VCP.write(buf) and USB_VCP.send(buf)

dimpolo, 2021-01-08:
What are the differences of these two methods?

Also what are the conditions for when these methods return something other the `buf.len()`

I thought I would ask here rather than on the forum, because I feel like the documentation could be improved here.

davehylands, 2021-01-08:
This is a bug tracker. The proper place to ask this is on the forum.

dimpolo, 2021-01-08:
I thoght incomplete and ambiguous documentation would warrant an issue. 
For example CPython does have issues about documentation in its "bug" tracker.
Sorry if this was out of place. Feel free to close this.

davehylands, 2021-01-08:
If you were submitting a PR to correct the documentation then this would be the correct place for it. But to me it sounded like you were asking the question "what is the difference between send and write".

I guess its a bit gray since you're talking about documentation rather than code.

I'm not sure what the exact difference is between send and write. If the buffers were close to full and you asked to do a send/write with a timeout, then it would return the number of bytes written when the timeout expires, which might not be the amount you asked to send (since the buffers filled up).

jimmo, 2021-01-11:
The difference is that `send` is handled directly by the VCP driver, whereas `write` is inherited from the VCP being a "stream object". This is a bit of a point of confusion generally on stream-like things (files, sockets, uarts, etc) -- read vs recv, send vs write, etc.

`write` calls usbd_cdc_tx_flow: Writes only the data that fits if flow & CTS, else writes all data. Returns number of bytes actually written to the device.

`send` calls usbd_cdc_tx: which will keep trying to write (up to the timeout), i.e. might involve waiting for CTS (potentially more than once).