micropython/micropython #6761: Can't mount SD card error

W-Erwin, 2021-01-09:
I am just starting with coding and bought a pyboard to make my projects. I want to use a SD card but when I insert it, it gives the error code 'Can't mount SD card'.

It does recognize the SD card, because when I use the code pyb.SDCard().present() it says it is True. I have tried a few different Sd cards but they all give the same problem. I tried formatting the Sd cards but that doesn't help either.

Does anyone know how I can solve this? Because to my knowledge it should be 'plug and play'. 
[image]

rolandvs, 2021-01-09:
This is more a question for the forum. However, I powered up one of my boards with NO SDCard inserted:
```
import pyb
pyb.SDCard().present() -> False
# insert card in slot
pyb.SDCard().present() -> True
# create a device
sd = pyb.SDCard()
sd.read(0) -> dumps a block (raw reading)
import os
os.mount(sd,"/sd1") -> create a drive
os.listdir("/sd1") -> show directory
['upysh.py', 'bckup.py', 'tsl2561.py',  'test.py', 'temp.py', 'templogger.py', 'servo.py', 'SensorBoards.py', 'rms.py', 'progress.py', 'pca9544.py', 'pca9544', 'mut.py', 'max44009.py', 'main.py', 'System Volume Information', 'logger.py','fusion.py', 'driver_pca9544.py', 'cli.py', 'boot.py', 'boarddac.py', 'board.py', 'bme280.py']
```

Also insert card and power up the system will boot from the SD Card inserted, if you have boot.py and main.py present and skirts the internal Flash.

Tested on a PYBv11 wih MPY 1.12.

robert-hh, 2021-01-09:
Micropython is a little bit picky about SD cards. The card must have a single FAT partition. Some cards have a hidden additional partition. So better reformat it before using. And not all models work. I had a good experience with 8, 16, and 32 GB cards from sandisk, transcend and samsung.

W-Erwin, 2021-01-10:
> 
> 
> Micropython is a little bit picky about SD cards. The card must have a single FAT partition. Some cards have a hidden additional partition. So better reformat it before using. And not all models work. I had a good experience with 8, 16, and 32 GB cards from sandisk, transcend and samsung.

I currently own a samsung 128 gb and a 2 gb SanDisk SD card, I will try and buy one of 8,16 or 32 GB and try again.

robert-hh, 2021-01-10:
The 2 GB model may work too. I did not mention it because I do not have one. Single partition FAT is also important.

dpgeorge, 2022-06-10:
Commit 63c7593df66533fc0c529d2ed2a5da38042c8970 allowed SD cards without a partition table to be mounted (on stm32, pyboard).  That may have fixed the problem.

I'll close this issue because it's quite old, but feel free to reopen if it's still not working.

closed 2022-06-10 by dpgeorge