micropython/micropython #6762: esp32: Check default SDCard pin assignments

jimmo, 2021-01-10:
From https://forum.micropython.org/viewtopic.php?f=18&t=9448&p=53524

I think machine.SDCard is supposed to default (for slot=3) to `SDSPI_SLOT_CONFIG_DEFAULT`

```
#define SDSPI_SLOT_CONFIG_DEFAULT() {\
    .gpio_miso = GPIO_NUM_2, \
    .gpio_mosi = GPIO_NUM_15, \
    .gpio_sck  = GPIO_NUM_14, \
    .gpio_cs   = GPIO_NUM_13, \
    .gpio_cd   = SDSPI_SLOT_NO_CD, \
    .gpio_wp   = SDSPI_SLOT_NO_WP, \
    .gpio_int  = GPIO_NUM_NC, \
    .dma_channel = 1 \
}
```

But https://docs.micropython.org/en/latest/library/machine.SDCard.html#esp32 says miso=12.

tve, 2021-01-19:
There's clearly a problem... For reference, the `SDSPI_SLOT_CONFIG_DEFAULT` def is in esp-idf `components/driver/include/driver/sdspi_host.h`. Docs state "For slot initialization, SDSPI_SLOT_CONFIG_DEFAULT can be used to fill in default pin mapping, which is the same as the pin mapping in SD mode." but I can't retrace how "is the same as" is to be interpreted...
My vote would be to remove the use of that macro and hard-code the definition. I don't have anything with an sdcard slot, so someone else should create a PR and test it.

jonnor, 2024-08-18:
This issue is now over 3 years old. Is this still problematic, or might it have been fixed by recent updates to esp-idf and the MicroPython esp32 port?

dpgeorge, 2024-08-19:
This was hopefully fixed with the update to IDF 5, see e4650125b88a35f074097f16d84a8f49bd22ac06

closed 2024-08-19 by dpgeorge