micropython/micropython #6764: read() function returns '\r' between lines.

HexVitor, 2021-01-12:
```
Board: DOIT ESP32 DEVKIT V1
CHIP: ESP-WROOM-32 (ESP32D0WDQ6)
Firmware: esp32-idf3-20200902-v1.13
SO: Windows 10 Home 20H2 64 bits
IDE: tested with Thonny 3.3.2 and VS Code 1.52.1
```

Hello everyone. I'm trying to read a file saved in the internal memory of my ESP32 for web communication purposes, but the **read() function of micropython is bringing '\r' after each line** of code, making it not work.
I tried to read the same file through my computer's Python interpreter and no '\r' was displayed.

Could someone help me with this? Is this a bug or should it happen?

## Procedure:

1. Create a file (eg 'test.txt') in the internal memory of ESP32;
2. Write and save the following on it:
```
Line1
Line2
Line3
```

3. Open the file using REPL:
```Python
>>> file = open('test.txt')
```

4. Read the file:
``` Python
>>> read = file.read()
```

5. Show the file:
```Python
>>> read
```
```
output: 'line1\r\nline2\r\nline3'
```

6. Save the same file to your computer;
7. Now read it through Python (same commands):
``` Python
>>> file = open('D:\test.txt')
>>> read = file.read()
>>> read
```
```
output: 'line1\nline2\nline3'
```

In both cases print(read) returns:
``` Python
Line1
Line2
Line3
```

Thanks for your attention.

dlech, 2021-01-12:
This is normal. Try `file.readline().strip()` to read one line at at a time.

peterhinch, 2021-01-12:
The line terminators in the file `test.txt` will depend on the OS used to create it. Windows uses `\r\n`, Linux uses `\n`. I've no experience of OSX but I'd guess that it's like Linux.

dpgeorge, 2021-04-14:
CPython has the concept of "universal newlines" and can convert `\r\n` into `\n`.  The docs state https://docs.python.org/3/library/functions.html#open :

> When reading input from the stream, if newline is None, universal newlines mode is enabled. Lines in the input can end in '\n', '\r', or '\r\n', and these are translated into '\n' before being returned to the caller.

MicroPython doesn't support this.  So you need to write code to deal with these kinds of newlines.

jonnor, 2024-08-18:
The behavior seems to be intentional, as noted above. So I suggest closing this issue.

closed 2024-08-18 by robert-hh